dronekit/dronekit-python #203: need CONTRIBUTING.md

mrpollo, 2015-07-15:
here are some topics we might need to cover, its up to you guys to decide what you want to see here @hamishwillee @tcr3dr 

- [ ] how to run locally
- [ ] test instructions
- [ ] code style
- [ ] descriptive commit messages

hamishwillee, 2015-07-15:
@mrpollo @tcr3dr 

This content should [go into the guide here](http://python.dronekit.io/contributing/index.html) and **CONTRIBUTING.md** should be a thin link to that guide.

I've modified the list above to include a few topics  we already cover, and to make it clear that there is a difference between running tests locally and rebuilding dronekit-python locally - though that might not be true.

Typically you'd also include some words about how we expect people to work with our master branch (we have) and who can merge. We might also have some words on how you can become someone able to merge changes. 

When the list is done, I can "drive" the docs. Would be good to have some bullet point docs from you guys about the first three test topics, and a brief discussion about what you want to see in each.

tcr3dr, 2015-07-17:
@hamishwillee I began a branch here that should fill in some of this info: https://github.com/diydrones/dronekit-python/pull/212

Shall we work out of this same PR to update the documentation quality enough to be merged?

hamishwillee, 2015-07-17:
@tcr3dr 

> I began a branch here that should fill in some of this info: #212
> Shall we work out of this same PR to update the documentation quality enough to be merged?

Yes. Good start. 
Unfortunately I'm not going to have time in the short term to look at this "properly". I have added some notes that I help though. My priority for the coming week will be getting a draft of the cloud quickstart and my ardupilot wiki editor role. 

In the short term just let me know if anything you write needs a review.