dronekit/dronekit-python #215: Travis doesn't allow forks from users to be tested (droneapi key)

tcr3dr, 2015-07-17:
Need a workflow that routes around this issue.

tcr3dr, 2015-07-17:
One solution is to not try the web-based tests on nonlocal forks, which is probably the most straightforward solution.

hamishwillee, 2015-07-17:
@tcr3dr What do the web tests cover that would be missed out? ie If we have tests then presumably they are important to run?

tcr3dr, 2015-07-22:
@hamishwillee I realize that tests are actually being run on Circle CI and Appveyor, and only Travis is excluded. So disabling web tests only for Travis still gives us full coverage.

What is actually of concern is that we have an API key that we expose in our tests. Unsure what to do about that other than to be vigilant.

hamishwillee, 2015-07-22:
@tcr3dr Good news.

> What is actually of concern is that we have an API key that we expose in our tests. Unsure what to do about that other than to be vigilant.

Bearing in mind that I know nothing about our infrastructure, some ideas

* Only allow requests from that key from certain IP addresses
* Run tests on mirrored test database, and ensure that key only has access to that database

tcr3dr, 2015-11-07:
This is mostly a PR hygeine issue. I don't think there is a solution except for running these tests locally, for now.

closed 2015-11-07 by tcr3dr