dronekit/dronekit-python #218: Vehicle system time for sensor syncing

djnugent, 2015-07-20:
Extract the autopilot time from incoming mavlink messages. It would be nice to correlate the vehicle state(attitude,location, etc) with a timestamp for use with other sensors. I plan to implement this for my own use but wanted some feedback from others for when I actually implement it.

djnugent, 2015-07-20:
Current implementation idea:
Vehicle.time() returns a running time from the vehicle. It is adjusted every time a *certain* message is received by the autopilot. The type of the *certain* message doesn't really matter. As long as it is updated regularly and has a timestamp, we are good. Like a location or attitude messages.

All vehicle attributes have a timestamp for when they were last updated(sent). Ex: vehicle.location.time or vehicle.attitude.time

tcr3dr, 2015-07-20:
I don't understand what `Vehicle.time()` would return, unless it's supposed to be passed an argument? In any case, I do like the per-attribute properties a lot.

We need to make sure to avoid race conditions for this API. If you read `vehicle.location`, then read `vehicle.location.time`, and the message were updated inbetween from another thread, that sounds undesirable. Perhaps we want a different type of API:

```
(value, update_time) = vehicle.location.state()
```

djnugent, 2015-07-20:
The `vehicle.time' would return the system time of the autopilot. It comes is handy when sending messages back to autopilot with timestamps. It would be nice if both the autopilot and droneapi were running a semi-synced clock. It could be an attribute or function call...it doesn't matter.

I like that idea. Long live tuples!!!

tcr3dr, 2015-07-20:
Got it. `Vehicle.last_timestamp` can be a bit more accurate, as it would imply it's not a synced clock, just a message update. And a clock diff tracker could be built on top of tracking onchange updates to that property.

djnugent, 2015-07-20:
No it would not be `last_timestamp`. It would be system time *shifted* to match the autopilot time.

Upon receiving a new message with a timestamp we would find the difference between the current system time and the timestamp. `shift = time.time() - msg.timestamp`
Then:
`Vehicle.time` returns `vehicle.time = time.time()  - shift`

tcr3dr, 2015-07-20:
Rather, let me suggest we have `Vehicle.last_timestamp` so that we could implement a simple `Vehicle.time()` on top of that primitive:

```py
def __init__(self):
    # ...
    vehicle._time_shift = 0
    vehicle.on_change('last_timestamp', lambda last: vehicle._time_shift = time.time() - last)

def time(self):
    return time.time() - self._time_shift
```

My reasoning is that `last_timestamp` is a sure bet as an API addition, but time synchronization is something that might need to be more complex: https://en.wikipedia.org/wiki/Clock_synchronization I'm not familiar enough to suggest time diffs are adequate enough for implementing `Vehicle.time()` or not.

djnugent, 2015-07-20:
I personally don't see any use for `last_timestamp` but that doesn't mean someone else wont. 

Yeah it isn't the most elegant solution for clock sync. I'm thinking about sitting in on tonight's dev call. Maybe the community will have some input on how to do clock syncing between mavlink enabled devices.

hamishwillee, 2015-07-20:
As attribute queries are synchronous and return after retrieving a messag,e the timestamp of the message seems irrelevant. If we were caching and then polling for last message update then I would understand more why we might need it. 

I think perhaps don't understand the use cases you're trying to address, and hence the solutions don't answer any question for me.

sahiljuneja, 2015-09-28:
Was this ever implemented? @hamishwillee @djnugent?

"It would be nice to correlate the vehicle state(attitude,location, etc) with a timestamp for use with other sensors."

hamishwillee, 2015-09-28:
@sahiljuneja Not yet. @tcr3dr Is this on roadmap for DKPY2.0 or ... 2.1?

sahiljuneja, 2015-10-10:
Hello @hamishwillee and @tcr3dr - I believe there are MAVLINK messages for the same that could be integrated with the code. I will look into that first as well.

hamishwillee, 2015-10-11:
@sahiljuneja This wasn't implemented. We can certainly get the messages, I just don't see the point (see https://github.com/dronekit/dronekit-python/issues/218#issuecomment-123093516). I doubt it will be in 2.0.

djnugent, 2015-10-13:
@hamishwillee Is it harmful to add this feature? It's obvious that this is not the ideal solution for time syncing but it is better than nothing. It should get you within 150-15ms time sync.

@tcr3dr I now understand your `last_timestamp` field. It makes sense as long as `vehicle.time` exists.

hamishwillee, 2015-10-13:
@djnugent I stick with my statement above:

I don't see any harm in it but nor do I see the benefit. As above:

> I think perhaps don't understand the use cases you're trying to address, and hence the solutions don't answer any question for me.

So do as you will!

alexblack, 2016-06-20:
Related to this - does anyone know how to find the timestamp of `vehicle.attitude`?

hamishwillee, 2016-06-20:
Do you mean attitude or altitude? Assuming the former, it looks like the [original message](http://mavlink.org/messages/common#ATTITUDE) includes a timestamp (time_boot_ms). YOu'd need to create your own message handler for this type to get the information though and do a little testing to verify that ardupilot actually sends it.

alexblack, 2016-06-20:
Thanks, yes I meant attitude.  Assuming ardupilot includes time_boot_ms that will be helpful.  I'm surprised DroneKit doesn't expose the timestamp.

hamishwillee, 2016-06-20:
>  I'm surprised DroneKit doesn't expose the timestamp.

I didn't make the decision, or question it. If there is quantifiable value in this information then we're more than happy to add the timestamp.

alexblack, 2016-06-20:
Without the timestamp you don't know how old the attitude data is.

Though, anyone who cares would probably use the message to receive
attitudes.

Having the timestamp data would have helped me understand what was going
on, for example I didn't realize it's only received at 4hz, if I was
sampling it faster than that then I'd have seen duplicate timestamps.

hamishwillee, 2016-06-20:
Well you know it was maximum of "4 khz" ago + the connection timeout value ... but I do see your point.

I'd be quite happy to have the information stored if we receive it. Possibly normalised so that readers could get a time since last received.

fnoop, 2017-02-08:
So I've had a quick go at this problem.  Created a class that keeps track of a timestamp in microseconds, with an estimate() method that attempts to estimate the flight controller time based on the last known time plus the time elapsed from the companion computer clock.
```
# Define tracking timer class
class TrackTimer:
    def __init__(self):
        self.fctimeus = None
        self.delta = None
        self.debug = False
    def update(self, timeus):
        if self.debug:
            print "Expected timeus: " +str(self.estimate()) +", actual timeus: " +str(timeus)
        # Messages come in or are processed asynchronously, out of order, so a later message may have an earlier timestamp.
        # So ignore any updates that are older than what we already know about
        if timeus > self.fctimeus:
            self.fctimeus = timeus
            self.mytimeus = datetime.now()
    def actual(self):
        return self.fctimeus
    def estimate(self):
        if self.fctimeus:
            self.delta = datetime.now() - self.mytimeus
            return int(self.fctimeus + (self.delta.total_seconds() * 1000))
        else:
            return None
```

You can then use a decorator/listener to try and grab timestamps from all incoming messages:
```
# Create a new tracktimer object to try and keep sync with the flight controller
trackfctime = TrackTimer()
# Setup a listener for all incoming mavlink messages and update our time tracker whenever possible
@vehicle.on_message("*")
def listener_all(self, name, message):
    try:
        trackfctime.update(message.time_usec)
    except:
        pass
    try:
        trackfctime.update(message.time_boot_ms * 1000)
    except:
        pass
```

At any time you can get an estimate:
```
print trackfctime.estimate()
```
Or the last known time:
```
print trackfctime.actual()
```

It looks like the listener often processes or returns messages out of order, so returns a message that has an older timestamp than a previously processed message.  So the class above ignores anything older than it already knows about, but this may not always be desirable behaviour.  I'm sure this can be refined better by others.