dronekit/dronekit-python #220: Add AUTO mode documentation

hamishwillee, 2015-07-21:
We have a few minor examples of using AUTO mode, but nothing exhaustive. It needs to show how to add mission waypoints, clear waypoints, use next. Also needs to show dynamic updating. 

One gotcha that needs to be tested is how vehicle.commands.next works/is implemented. #201 research seems to indicate that next is only switched over once the current command completes - if you need to switch immediately then you'd use https://pixhawk.ethz.ch/mavlink/#MISSION_SET_CURRENT . Both ideas need to be tested. 

Also need to make it clear that you don't need to set the HOME position (waypont 0).

tcr3dr, 2015-07-22:
My only caveat here is we need to keep the right level of abstraction. AUTO mode for Solo is increasingly trending toward GUIDED + refreshing of waypoints. I anticipate DroneKit will be supporting both for some time but we want to avoid when possible the need to memorize two conceptual models.

hamishwillee, 2015-07-22:
@tcr3dr Can you expand on what you mean by "Guided" + refreshing of waypoints? My guess is that you're saying missions are now done in guided mode using scripts rather than as "mission definitions" in AUTO?

Appreciate that we don't want to confuse users. Using AUTO is important at the moment mainly because there are plenty of functions only available in auto mode. In Plane particularly this is so bad that the "guided" model is actually to dynamically modify a mission. 

We really need to get our basic guided API "up to scratch" (like addition of change speed and roi commands in interface) with corresponding changes in plane/copter to support them. At that point AUTO mode could really fade away.