dronekit/dronekit-python #222: self.vehicle.parameters is not iterable in a friendly way

tcr3dr, 2015-07-21:
```
for i in self.vehicle.parameters:
           logger.log(i)
```

On one setup this caused a keyerror, due to it iterating in sequence on the underlying object:

```
File "/usr/lib/python2.7/site-packages/droneapi/module/api.py", line 29, in __getitem__
   return self.__module.mav_param[name]
KeyError: 0
```

I haven't reproduced this yet.

hamishwillee, 2015-07-21:
@tcr3dr Just FMI, does/will this mean that you can iterate Vehicle.parameters and list all the parameter names and current values? (if so I will need to add this to the docs - it never occurred to me you might be able to!)

tcr3dr, 2015-07-21:
@hamishwillee It would be desirable! I haven't dug into how it is implemented that the `parameters` object doesn't act like a normal dict, but that should be cleared up.

OTOH, this is more an enhancement than a bug, so marking appropriately.

hamishwillee, 2015-07-21:
@tcr3dr Yes it would be desirable. Sounds good.  Lets take a note to create a docs issue when this resolves - if needed.

tcr3dr, 2015-07-22:
@hamishwillee Sounds good.