dronekit/dronekit-python #230: vehicle.commands must be reset after clearing

hamishwillee, 2015-07-23:
If you clear `Vehicle.commands` then you must redownload the mission before trying to add any new  waypoints or the addition will fail. The code below demonstrates this - if you remove the commented out lines then the new waypoint is added, but otherwise it is not.

This feels like a bug. If you've just cleared the commands from the mission then you're expecting it to be empty and ready for re-use. Even if the clearing failed at the autopilot end I would still hope that when I upload my mission it will be doing so assuming a cleared mission list.

```
"""
Tests resetting the vehicle.commands object
"""

import time
import math
from droneapi.lib import VehicleMode, Location, Command
from pymavlink import mavutil

api = local_connect()
vehicle = api.get_vehicles()[0]
cmds = vehicle.commands

def currentmission():
    """
    Print the current mission
    """
    print "current mission:"
    cmds.download()
    cmds.wait_valid()
    for cmd in cmds:
	    print cmd

print 'CLEAR MISSION '
cmds.download()
cmds.wait_valid()
cmds.clear()
vehicle.flush()

print 'Reset the cmds object'
#cmds.download()
#cmds.wait_valid()

print 'Add dummy waypoint'
cmds.add(Command( 0, 0, 0, mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT, 
                mavutil.mavlink.MAV_CMD_NAV_WAYPOINT, 0, 0, 
                0, 0, 0, 0, 10, 10, 10))
vehicle.flush() # Send commands

time.sleep(1)

print 'Check we have just 2 items in mission.'
currentmission()
```

P.S. Note the time.sleep near the end. This is required for the reasons given in  #227. You can remove it and everything after it and still verify this issue by loading the waypoints using mission planner.

hamishwillee, 2015-07-28:
Actually I am not sure this is  valid - though there is some sort of bug. If I split the commands to clear and the command to add a waypoint into separate scripts then the whole ree-download thing is not needed to add new waypoints.

I suspect the above script requires the meission be re-downloaded due to some sort of timing issues.

```
"""
Tests clearing vehicle.commands object
"""

from droneapi.lib import Command
from pymavlink import mavutil

print 'CLEAR MISSION '
api = local_connect()
vehicle = api.get_vehicles()[0]
cmds = vehicle.commands
vehicle.commands.clear()
vehicle.flush()
```

```
"""
Tests adding a command
"""

import time
from droneapi.lib import Location, Command
from pymavlink import mavutil

api = local_connect()
vehicle = api.get_vehicles()[0]
cmds = vehicle.commands

def currentmission():
    """
    Print the current mission
    """
    print "current mission:"
    cmds.download()
    cmds.wait_valid()
    for cmd in cmds:
        print cmd

	
print 'Add dummy waypoint'
cmds.add(Command( 0, 0, 0, mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT, 
                mavutil.mavlink.MAV_CMD_NAV_WAYPOINT, 0, 0, 
                0, 0, 0, 0, 10, 10, 10))
vehicle.flush() # Send commands

time.sleep(1)

print 'COUNT0: %s' % cmds.count	
print 'CURRENT MISSION'
currentmission()
```

hamishwillee, 2015-07-28:
Hi @tcr3dr

I understand what is going on now.

When you clear a mission you create an empty ``vehicle.commands`` object. However on the vehicle there is a mission command for the home location, and if you were to download the mission you would end up with one item in the cmd. 

When you upload the new (truly) empty cmd you run into  #132 - the first command is thrown away. So as things are, if you call clear, you immediately need to re-download the waypoints so that you start in a "good condition" for working with the cleared CommandSequence.

hamishwillee, 2015-11-02:
This is fixed by #384

tcr3dr, 2015-11-07:
This should also be tested by test_waypoints.py.

closed 2015-11-07 by tcr3dr