dronekit/dronekit-python #234: Add support for IMU raw values

albertplawinski, 2015-07-24:
Can you add support so that the API the query the raw IMU on board for the python code to interpret?

hamishwillee, 2015-07-27:
Thanks @albertplawinski . The packets are sent so there is no reason why not. Feel free to add a patch to support these yourself if you need them urgently as this is not likely to be an urgent priority.

@tcr3dr Copter and Plane support three IMU messages (MSG_RAW_IMU1 - MSG_RAW_IMU3) which I am guessing are split up versions of [RAW_IMU](https://pixhawk.ethz.ch/mavlink/#RAW_IMU) in the protocol? Rover omits the #RAW_IMU2. Currently we support "not supported" by returning None for a call to an object which can't be "satisifed". We should perhaps also think about implementing the capability API in Python (which I would hope would tell us what we can and cannot do on a particular platform?)

tcr3dr, 2015-07-28:
@albertplawinski You can prototype this with http://python.dronekit.io/automodule.html#droneapi.lib.Vehicle.set_mavlink_callback in your own scripts. If you have a proof-of-concept, post it here, as it will increase the likelihood of it getting added as a feature sooner.

@hamishwillee Made a note on https://github.com/dronekit/dronekit-python/issues/250.

hamishwillee, 2015-07-28:
@tcr3dr Thanks. I hope to write a topic on how use set_mavlink_callback for this sort of thing. I was thinking along the lines of "Supporting New MAVLink Messages" with suggestion of the "right way" being to create issue and follow our contribution process, but that for temporary relieve use this API ....

hamishwillee, 2015-08-10:
@tcr3dr @mrpollo I am thinking of implementing this API as part of writing a guide topic on how to implement an attribute to support a new message type. This would work with #277.

1. Good idea?
2. This is quite easy to do as Vehicle.raw_imu but we were discussing having larger abstractions of the API so as not to clutter the Vehicle namespace. So for example we might have ``Vehicle.imu`` which holds ``vehicle.imu.raw_imu`` and ``vehicle.imu.scaled_imu``

  * What level of abstraction do we have in mind - for this case, and for the broader case. We probably need to look at sets of messages to see what makes sense (so I can offer advice in the guide)

3. Currently attribute parameters are dumped as members of the [main module like this](https://github.com/dronekit/dronekit-python/blob/master/droneapi/module/api.py#L358). I would think this might lead to namespace clashes later - which is why having separate classes for things like HUD message might be good guidance (ie you can get speed from a number of places, and we don't want use self.speed for both.

4. Are there any other considerations I should be aware of/guidance I should provide?