dronekit/dronekit-python #251: DRONEKIT_PATH variable

djnugent, 2015-07-28:
Currently dronekit only supports relative and absolute paths. It would be nice if we could point `api start <file>` to look at a set of directories when running a script. DRONEKIT_PATH would point to your project folders.

This idea was brought up in a recent dronekit session Chris and I had. At a minimum we should specify in the documentation that if that user can't specify the directory in which mavproxy is started(i.e. running SITL), then they must start use an absolute path to call their script.

DRONEKIT_PATH should not be necessary for when we run dronekit without mavproxy. But at a minimum we should document that the user should provide an absolute path when running dronekit scripts.

zlite, 2015-07-28:
Also, let's update the documentation to show people how to download the
example files:

git clone https://github.com/dronekit/dronekit-python/examples

hamishwillee, 2015-07-28:
Yes, it is painful to have to enter full search paths - I'm always thinking "what is my current path" and I tend to just close the prompt and reopen it in the new folder I want to run. I like the idea, but I don't think we'll need it when we have dronekit running as a module, so I guess whether we do this depends on priority/effort vs how quickly we'll get dronekit running as an independent module. @tcr3dr do you have a view on that?

@djnugent  @zlite Regarding the docs - In both cases the "missing" documentation is present (see below).

> But at a minimum we should document that the user should provide an absolute path when running dronekit scripts.

We do document this [here in the getting started](http://python.dronekit.io/guide/getting_started.html#running-an-app-example):
[image]

> Also, let's update the documentation to show people how to download the
example files: ```git clone https://github.com/dronekit/dronekit-python/examples```

We do explain how to get the examples in [Running the Examples](http://python.dronekit.io/examples/running_examples.html). 

[image]

You can't AFAIK just clone the example tree as you've shown, so I demonstrate how to clone the whole branch.

I am however missing this information in the getting started (http://python.dronekit.io/guide/getting_started.html#running-an-app-example) so perhaps the solution is to add how to get the examples there?

I am not sure how to better highlight the absolute path thing, so advice welcome ...

zlite, 2015-07-29:
Thanks, Hamish. I think this is more an issue of documentation
organization, rather than totally missing information.

For example,  for my OSX/SITL runthrough, I started at the Getting Started
<http://python.dronekit.io/guide/getting_started.html> page (reasonably
enough!).  It took me to the "Running an app/example" section, which is
where I ran into trouble because I didn't know how to download the example
script into my Vagrant instance. And even once I figured out how to copy it
from my OSX drive to Vagrant, I didn't have it in the right directory.

As you point out, all these questions are answered in the "Running the
Examples <http://python.dronekit.io/examples/running_examples.html>" page,
but that wasn't linked on the Getting Started page so I didn't see it.

So I'd suggest either embedding the Running the Examples info in the
Getting Started page, as you recommend, or at least linking to that page
with a highlighted note that you have to go through those processes before
going further in the tutorial.

Needless to say, these documentation issues are just a symptom of a
too-complicated toolchain, so I'm hoping Dronekit 2.0 solves this at the
root-cause level.

-Chris

hamishwillee, 2015-07-30:
Hi Chris / @zlite 

I've updated the documentation - you can see the changes in #261. The documentation in both sections is now much better, and also very similar. The only real difference being that the getting started focuses on a specific example. 

In both cases they start with how to get the examples/where the examples are located.

> Needless to say, these documentation issues are just a symptom of a too-complicated toolchain, so I'm hoping Dronekit 2.0 solves this at the root-cause level.

Very charitable of you - sadly I think I just needed to spend a bit more time on this section. 

A  lot of really good work has been done on the toolchain in the last few weeks.

* @stephendade  finished a new windows DK installer that runs over the top of the MAVProxy installer (my next task to document) 
* @tcr3dr and fiends have nearly finished dronekit-sitl-runner - a script which automatically downloads and sets up SITL for you without any of the Vagrant stuff ... on all platforms. 

Yes, that should make this much more bulletproof!

hamishwillee, 2015-07-30:
@tcr3dr the docs are updated, but we still need to decide if it is worth creating better handling for file management. My gut feeling is that the speed of progress on making DK a stand alone module means that this is not worth doing.

Open or close?

tcr3dr, 2015-10-14:
Looks like close as of DKPY2. Recommendation is to use standard Python path tools.

closed 2015-10-14 by tcr3dr