dronekit/dronekit-python #252: Expose home location as separate from .commands array

tcr3dr, 2015-07-28:
#230 #207 #132 

Home location doesn't act in the same way as our .commands array (see above topic for discussion). Instead of patching unexpected behavior, we can extract this "initial" command as a separate `Vehicle.home_location` variable.

hamishwillee, 2015-07-28:
The attribute should also be settable (the [MAV_CMD_DO_SET_HOME](http://copter.ardupilot.com/wiki/common-planning-a-mission-with-waypoints-and-events/common-mavlink-mission-command-messages-mav_cmd/#mav_cmd_do_set_home) command can be set any time)

FYI, the ardupilot implementation of home location as first waypoint is a "bug" that they hope to undo at some point.

hamishwillee, 2015-11-02:
This is fixed by #384

tcr3dr, 2015-11-10:
Complete!

closed 2015-11-10 by tcr3dr