dronekit/dronekit-python #258: http://python.dronekit.io/about/release_notes.html is wrong about patch releases

tcr3dr, 2015-07-30:
So I've inadvertently been documenting patch releases, unlike what it reads on http://python.dronekit.io/about/release_notes.html:

> Patch releases are not separately documented here.

Should we only publish release notes for minor/major branches? Should that line be changed? Or should bugfixes be rolled up into the release notes for the minor branch and published to the site? Or should all release notes be published there?

hamishwillee, 2015-07-30:
@tcr3dr @mrpollo The regular updates of point release notes are useful - they show exactly what has changed and help people understand whether updating makes sense. 

That said, I'm not sure they are the right thing for the release notes page. IMO the page is an opportunity for us to show off all the great new things we've been doing and that this release delivers. The point releases are too small to do this.

I *think* a rolled up release notes that captures all the changes in the current minor release up to the current pip release would be good (and this would roll with each pip release). So in this case a release note made up of everything from notes in 1.4.0 - 1.4.x.

Of course depends on how much effort. This wouldn't be my highest priority right now.

P.S. Arguably we also need a high level hand crafted note on what the API delivers in the current major version - API coverage etc. However that might be worth kicking off in 2.0 when we change the architecture.

tcr3dr, 2015-07-30:
Totally think it's worth taking through, since I feel the same way when authoring the updates. My questions:

* Should the Github release page show only the diff or the cumulative release notes?
* The docs should show cumulative data for the minor or major branch. Should it do it with all bugs, features, etc. grouped together, and calling out specific additions by which point version they were added in? Or should they do it segmented (i.e. at the bottom of the release notes)

I'm fine with either, it takes about the same amount of time to author notes. Would be nice to get this in place sooner so it's a no-brainer by 2.0.0.

hamishwillee, 2015-07-30:
@tcr3dr I think that for docs it would be good for bugs etc to be grouped together with each line pre-fixed with the point version. So in each release you'd append the bugs for that release to the list.

Ideally we'd still link to the "latest point release" docs on github, in the same way you have them now. 

That all said, I'm having trouble envisaging what each case would look like. Can we mock up somehow?