dronekit/dronekit-python #269: Feature Request:  Access to Mavlink messages received by Autopilot

a7scenario, 2015-08-03:
For Companion Computers (e.g. Raspberry Pi) running DroneKit onboard an aircraft, it's easy to access outgoing mavlink messages that the autopilot sends to the GCS.  You can do this by simply invoking DroneKit's set_mavlink_callback() function.  However, there doesn't appear to be any way in DroneKit to intercept mavlink messages going in the other direction -- messages that the GCS sends to the Autopilot (e.g. MAV_CMD_DO_SET_MODE). 

This is particularly limiting for mavlink messages that the autopilot doesn't ACK, as there doesn't seem to be any way for the Companion Computer to know if a specific message was sent from the GCS to the Autopilot.

hamishwillee, 2015-10-05:
@a7scenario My take on this is that the companion computer only gets messages routed to it - so if the GCS sends a specific message to the autopilot that doesn't somehow go via the companion computer there is nothing for dronekit to receive. So what I think you want is some way to ensure that messages from GCS to ArduPilot are routed to the Companion computer. Is that correct?

a7scenario, 2015-10-05:
Yes, exactly.

a7scenario, 2015-10-05:
Uploaded a graphic showing the behavior I'm seeing:   http://justpaste.it/o53o

hamishwillee, 2015-10-05:
@rmackay - is this do-able - ie can we auto-route all messages to the companion computer so that dk can choose to ignore them if it wants?

hamishwillee, 2015-10-07:
From Tridge:

> @Chambana if the companion computer uses the same sysid as the autopilot, but a different component ID, and if the GCS sends the messages to the broadcast component ID (ie. zero) then the messages should be automatically forwarded to the companion computer

@tcr3dr Most of this is up to the GCS. But to support this model we need to ensure that dronekit uses the same sysid as the autopilot but a different component ID. We either need to make this happen automatically (if it doesn't already) or provide an API/mechanism to allow them to be set.

AfshinMr, 2018-11-01:
@a7scenario May I ask if your problem is solved>? Could you explain it to me if it is solved?
Thank you, I appreciate it.