dronekit/dronekit-python #270: Companion Computer sending of arbitrary data/files over 3DR telem radio

a7scenario, 2015-08-03:
A send_data() function in DroneKit would be very useful -- allowing for onboard Companion Computers (e.g. Raspberry Pi) to transmit arbitrary data files back to the GCS over the 3DR telemetry radio link.  For a lot of lower-bandwidth drone applications, this would help us avoid having to run a 2nd radio link on the aircraft (Wi-Fi or GSM).  

Perhaps the DroneKit could write data chunks (ENCAPSULATED_DATA messages?) to the local pixhawk and fill in the message's sysID/componentID's with the GCS's sysID/compID -- possibly forcing the pixhawk to route/relay the data message(s) to the GCS over the 3DR telemetry link?  On the ground side, QGC and the new APM planner 2, if I'm not mistaken, allow for local UDP relay of any incoming data from the 3DR telemetry radio -- which would allow you reconstruct the file on the ground and write it to disk.  

It might be complicated to handle ACKs due to Issue #269, unless the radio link handles that already.  Not sure!

hamishwillee, 2015-08-04:
+1 Sounds like a great feature.

a7scenario, 2015-08-07:
It looks like its possible to route arbitrary data over the 3DR telemetry radio down to your Ground Station from a companion computer running DroneKit.  After message creation in message_factory (entering your GCS as target_system), the key is to not invoke DroneKit’s send_mavlink() function -- because it changes the target_system/target_component to your autopilot’s IDs (and you want to send to the ground/GCS).  Instead, you have to access a name-mangled superprivate class variable in MPVehicle and directly write the msg to the autopilot if you’re running a DroneKit script.  
So, instead of using:
    vehicle.send_mavlink(msg) 
use this:
    vehicle._MPVehicle__module.master.mav.send(msg)

Proof of concept code:
     https://gist.github.com/a7scenario/2ca4f0f99087a35aca00

A file_transfer_protocol mavlink message with your data payload should pop in your GCS, having been sent over the 915mhz 3DR telemetry radio.  This doesn't address ACK'ing the transfer nor relaying the messages out of the GCS software to a local UDP listener on the ground station, which certainly makes this a bit complicated :)

hamishwillee, 2015-08-09:
@tcr3dr Interesting discussion. I think this highlights the need to be able to send messages to the GCS from the companion computer. Possibly the simplest way to handle this is is to update send_mavlink so that  it treats 0 as autopilot component/system id but otherwise respects values set by the user?

The actual handling of acknowledgement etc is something I believe we're thinking about in another issue.