dronekit/dronekit-python #276: Vehicle.flush() not working as intended

jamesbellaero, 2015-08-07:
There are two issues with the flush method that I have found:

1. When I add waypoints and call flush(), the entire mission requires about .25 seconds per waypoint to write in lab conditions. If I attempt to set the vehicle to auto or download the waypoints before all of them are written, the Pixhawk will not receive the full mission and 're-request' the waypoints it missed. When flying, this caused the quad to crash since it got set to auto without home being written properly and came down. 
To get around this issue, I call time.sleep(vehicle.commands.count/2) immediately after calling vehicle.flush(). However, if flush() worked properly it would automatically wait until receiving confirmation from the quad that all waypoints had been written before continuing, much like wait_valid() does. 

2. After downloading waypoints from the Pixhawk with vehicle.commands.download() and vehicle.commands.wait_valid(), the Pixhawk no longer has a loaded mission, meaning that I can't set it to auto while armed, etc. To reupload the mission I have to actually change a command in the CommandSequence. 
I would have figured that simply calling flush() would handle the reupload without me having to change anything, but it doesn't.

These are important issues which need to be solved before lots of people start using Dronekit in real applications. Either fix the flush method add new methods to achieve the required functionality.

hamishwillee, 2015-08-09:
@jamesbellaero Thanks for this. It is a duplicate of #227.

@tcr3dr I agree with @jamesbellaero that this is a critical defect. It has certainly cost me a LOT of time when writing examples. I am not sure of the correct solution - do you think we could get some time on it in the coming week?

tcr3dr, 2015-11-09:
See #227 and #439 for a test case that should indicate that upload completes without issue, as well as re-downloading commands.

hamishwillee, 2015-11-10:
@jamesbellaero As @tcr3dr says above, we think this is fixed in dkpy2.0. Can you please confirm?

jamesbellaero, 2015-11-10:
I'm not currently in the position to do so as I'm back in school from my
internship. Once we get our test setup completed in my student group I'll
let you know.

tcr3dr, 2015-11-10:
Thanks @jamesbellaero. When you're able to recreate it, feel free to re-open this issue or submit a new one—we are committed to ironing out the bugs in DroneKit Python for this next release and would love your input.

closed 2015-11-10 by tcr3dr