dronekit/dronekit-python #277: Add improved example for set_mavlink_callback

hamishwillee, 2015-08-10:
``set_mavlink_callback`` allows users to intercept all messages to the companion computer/script, so it is arguably* quite useful for adding support for messages or "attributes" that aren't yet in the API. The proposal is to update the example to provide a "useful" reference for people who want to create their own message handlers.

* I say "arguably" because it is similar effort to update the API to do this properly. So you would only use this while waiting on changes to go into the API "properly". The example would co-exist with an enhancement to the contribution section providing step-by step guidance on how to support a new attribute. The instructions would make it clear that the best approach is to support the attribute.

hamishwillee, 2015-08-10:
@tcr3dr @mrpollo @djnugent  I have created an example ([see gist here](https://gist.github.com/hamishwillee/1ff5c3771eea307473c0#file-handle_message-py)) which shows how you might use ``set_mavlink_callback`` to the raw_imu message. The example is a little non-trivial in that it attaches the new object ``vehicle.raw_imu`` to the vehicle and hooks into the normal notifier mechanism so you can create callbacks and, after adding the handler, treat this as though it were an inbuilt attribute.

I would have liked to make this a separate module but there are two problems

* I don't think you can load a module/file that isn't properly in the path - ie you can't import a file "in the same directory" because the current directory is where MAVProxy starts, not where your first file is run from. Yes we could do some magic to add paths but not worth it here unless there ends up being an API to easily set current path from MAVProxy.
* ``set_mavlink_callback`` is a "singleton" - if I hook it in my module, then anyone adding a new observer in their script would break it.

I quite like this, because it is probably a more realistic way you might use the API and because we're not too fast to add new attributes. I dislike it because I would rather encourage people to add new APIs properly.

A question

1. Is this a reasonable example, or should I just do a basic one which simply accesses the values of the received message?

tcr3dr, 2015-08-11:
I think this mechanism, though less than ideal, will actually go a fair way toward helping us shape what the ideal API looks like for implementing these attributes. So in the case that you add your own parsed messages from MAVLink and then use them in the same example, this is the most common usage of `set_mavlink_callback` even if we don't recommend it—creating globals or modularizing it is simply not possible at this time. When we extend the API though, we can naturally update the same code and hope implementations that used the old methods (which will continue to work) follow.

akwatson, 2015-08-11:
@hamishwillee I've noticed this as well and thought my linux path was just messed up (and it probably is...).  For one, try adding sys.path.append(os.getcwd()) to your primary DroneKit script launched via "api start" in Mavproxy and it will sometimes resolve the issue you mention above by grabbing/adding the local scripts' paths to the system path on the fly (instead of using mavproxy's path as the local path I think).  That seems to somewhat address the local import issue caused by running within Mavproxy, however, it seems DroneKit still chokes for me if my local imports contain classes that handles the DroneKit init stuff.  Would love to see add'l documentation on how to properly import several "helper" DroneKit scripts when launching a DroneKit script from the Mavproxy environment, assuming that this is even a legit issue and not just my system path being messed up :).  At the moment, I'm still having to copy/paste a lot of my DK classes into a single file, which can't be right!

"I don't think you can load a module/file that isn't properly in the path - ie you can't import a file "in the same directory" because the current directory is where MAVProxy starts, not where your first file is run from. Yes we could do some magic to add paths but not worth it here unless there ends up being an API to easily set current path from MAVProxy."

hamishwillee, 2015-08-11:
@akwatson Thanks. If adding sys.path.append(os.getcwd()) allows you to import files in the current directory that would be excellent. If it is unreliable though, I won't document it.

We're expecting this to be a null issue in the next major release because DK will become an ordinary module (not run in the MAVProxy context) - see #188. We have also discussed an interim solution (#251) but I personally think that #188 will deliver first rendering it not-needed. Until one of them delivers your only real choices are to always launch MAVProxy from the directory containing (all) the modules OR to make it a proper module installed using PIP/whatever. 

@tcr3dr Thanks very much. Arguably if #188 delivers and we allow multiple set_mavlink_callback observers this would be a better way to extend the API than modifying the main module. I will proceed with this as-is.

eliao, 2015-08-11:
I use 'sys.path.append(os.path.realpath(''))' in my dronekit scripts to allow them to import each other no matter where they're run from.

a7scenario, 2015-08-11:
@eliao I've been using this for helper functions, but it never works for importing my primary DroneKit class (which calls local_connect, api.get_vehicles(), etc).  Are you successfully able to run a DroneKit script that imports a local module (using the sys.path.append command above) and have that imported module call local_connect and api.get_vehicles without failing?  If so, I think that this path issue may be made moot by using the sys.path.append function.

eliao, 2015-08-11:
Oh, interesting.  Perhaps you are right regarding the main class.  Odd.  Maybe it's how Mavproxy loads up the initial module?

tcr3dr, 2015-11-07:
@hamishwillee We'll need a strategy for this now that we changed the callback mechanism for attributes and messages.

hamishwillee, 2015-11-07:
@tcr3dr OK, so the create_attribute example has been updated to the new attributes/messages. However it still assumes that you're using this in your script as a temporary measure while you wait for proper support in the API.

So the original purpose of the issue is addressed and this can be closed.

But by strategy, do you mean a way for people to write message parsers that can be attached to the running script, allowing a separate library to be added? It would be quite nice to be able to create a separate parser file for each attribute and add/remove support just by adding/removing a file from a particular place.

tcr3dr, 2015-11-09:
@hamishwillee I do like the general mechanism / idea, but it's probably too "early" to explore doing that in a composable way. Going to defer to #437 for one mechanism which might work (in a single-inheritance class system) but otherwise not try to explore more flexible methods yet.

closed 2015-11-09 by tcr3dr

hamishwillee, 2015-11-09:
@tcr3dr Absolutely agree, and love the approach you've done in #437. Thanks.