dronekit/dronekit-python #278: VehicleMode should be an enum

atomictom, 2015-08-10:
VehicleMode should be an enum or at least enum-like. The biggest issue with the current implementation is:
```python
VehicleMode("GUIDED") == VehicleMode("GUIDED") is False
```
This means that determining which mode you are in either requires you to make your own enum:
```python
GUIDED = VehicleMode("GUIDED")
LAND = VehicleMode("LAND")
```
and only use those variables. 

Test for string equality:
```python
str(vehicle.mode) == "VehicleMode:GUIDED"`
````
Or access the `name` field (which is not documented):
```python
vehicle.mode.name == "GUIDED"
```

Additionally, being an enum would allow for code that checks whether a given mode exists programmatically:
```python
if "LAND" not in VehicleMode:
    # do something
```
and to iterate through the available enums:
```python
for mode in VehicleMode:
    menu.add(mode)
```

The type of enum I have in mind is detailed in [this StackOverflow post](http://stackoverflow.com/questions/36932/how-can-i-represent-an-enum-in-python)

hamishwillee, 2015-08-10:
Hi @atomictom 

I absolutely agree that we should be easy to test for equality, to iterate through the modes, and to test for existence as you do above. I'm not sure that an enum is the right way to do this - probably this should be handled similarly to parameters (except they need to be observable too). 

I have therefore changed the title of this issue.

Regards
Hamish

P.S. [VehicleMode.name](http://python.dronekit.io/automodule.html#droneapi.lib.VehicleMode.name) is documented, and this would be the right way for comparing for equality.

atomictom, 2015-08-12:
An enum was the first thing that came to mind, but anything that filled those requirements would work fine. One additional thought I had was, it would be nice if there were only one reference to a VehicleMode. That way, you could test for equality with the `is` operator, perhaps like so:
```python
if vehicle.mode is VehicleMode.GUIDED:
    # do stuff
```

Also, you're right! Perhaps if there was a code example in the docs that used it (perhaps in an if-statement):
```python
# Test to see if we are in GUIDED mode
if vehicle.mode.name == "GUIDED":
    vehicle.armed = True
```
Thanks,
Thomas

hamishwillee, 2015-08-14:
@atomictom The code examples do use it - see [Set attributes](http://python.dronekit.io/guide/vehicle_state_and_parameters.html#vehicle-state-set-attributes). 

To make this more clear I have added a small section to the VehicleMode API reference - see #278. I hope you think that is sufficient for now.

hamishwillee, 2015-08-14:
I've spawned off #289. I don't think we actually want the mode iterable, but we do want to be able to get the modes, if only to test that the mode we want to change to is supported.

atomictom, 2015-08-14:
> and I can't see why Vehicle.mode would not be better just as a string.

I agree there. A string would be sufficient as far as I can tell. However, if there is some reason for having a class, another possibility is to implement the `__eq__` magic method.

> I hope you think that is sufficient for now.

Something like that is what I was looking for, so that works. Thanks!

> I don't think we actually want the mode iterable

Do you mean a single mode or all the modes? I do not personally have a use-case for it, but having an iterable of all the modes allows the available modes to be dynamically discovered, which could be useful to someone.

mrpollo, 2015-08-14:
I submitted a fix for this, it's annoying and IMO non-pythonic to not be able to compare objects, there might be other objects we need to compare aside from just `VehicleMode` like `Vehicle` maybe a good feature to have later when we are dealing with multiple `Vehicles`.

Thanks for the report @atomictom, this should go into the next minor release (no ETA for that yet)

hamishwillee, 2015-08-14:
@mrpollo 's fix is at https://github.com/dronekit/dronekit-python/pull/292

atomictom, 2015-08-14:
No problem! Thanks for the fix @mrpollo and the comments @hamishwillee.

commit 78dccb36 (2015-10-22): VehicleMode is now comparable

fixes #278
PR #292 (2015-10-23): VehicleMode is now comparable
this change makes it safe to compare `VehicleMode` objects with other `VehicleMode` objects and with strings

```python
vehicle.mode == VehicleMode('GUIDED')
vehicle.mode == 'GUIDED' 
```

```python
vehicle.mode != VehicleMode('GUIDED')
vehicle.mode != 'GUIDED' 
```

fixes #278
closed 2015-10-23 by tcr3dr