dronekit/dronekit-python #285: condition_yaw doesn't work without a call to send_global_velocity

atomictom, 2015-08-13:
I was testing out the `condition_yaw` in the docs and it while the command would send a MAVLink message, and the drone (SITL) would respond with success, the drone would not yaw (as indicated by the `vehicle.attitude.yaw` field). However, if I precede yaw commands with a `send_global_velocity(0, 0, 0)` command (again, taken from the docs), the drone will rotate successfully. 

I am executing these commands in GUIDED mode. As far as I can tell, there doesn't seem to be a mention of this requirement in the docs (I guessed at it since the yaw examples included `send_global_velocity`).

hamishwillee, 2015-08-14:
@atomictom Can you create a gist of your example code please - ideally based off http://python.dronekit.io/examples/guided-set-speed-yaw-demo.html ?

atomictom, 2015-08-14:
https://gist.github.com/atomictom/9bae5cb7642045b184da

It uses mostly unmodified versions of the `condition_yaw` and `send_global_velocity` functions. If you uncomment either of the lines with `# send_global_velocity(0, 0, 0)` then it yaws, otherwise it won't.

I load it up using `mavproxy.py --master=172.28.128.1:5760 --cmd="module load droneapi.module.api ; api start gist.py"` connecting to a vagrant version of SITL. After you see the confirmation `Got MAVLink msg: COMMAND_ACK {command : 115, result : 0}` the yaw should (when working correctly) change significantly.

hamishwillee, 2015-08-18:
The gist didn't work (missing yaw() ) but I modified it and yes, I think you are correct. You can of coruse just move to your current position, but it is a bit annoying. This will be autopilot behaviour, not dronekit as such, but yes we should document it. 

@rmackay9 - is this a known issue with Copter, and can we do anything about it?

atomictom, 2015-08-18:
Ah, sorry about that. I renamed `yaw()` to `condition_yaw()` just before posting it to be consistent with the name in the docs. [Here is an updated version](https://gist.github.com/atomictom/b10fd1af66443ddbe1fb) for anyone who is interested.

atomictom, 2015-08-18:
Also, I'm sorry, I did not mean to close it...

eliao, 2015-08-18:
I believe @asetapen encountered this yesterday and has some additional details.  Apparently this only happens after using a guided takeoff.

hamishwillee, 2015-09-08:
This will be resolved by https://github.com/dronekit/dronekit-python/pull/307

asetapen, 2015-09-09:
Woohoo! Yes I've seen this, thank you @hamishwillee for clarifying.

hamishwillee, 2015-09-09:
@asetapen NO worries. I'd hoped @rmackay9 would consider it a bug, but 'twas not to be.

tcr3dr, 2015-11-09:
Resolved by #307.

asetapen, 2015-11-09:
@tcr3dr++, @hamishwillee++!

hamishwillee, 2015-11-09:
Well, it isn't *really* resolved - just "noted as an issue :-) We could choose to send a dummy movement command first, but I think what we do now is "safer".

tcr3dr, 2015-11-09:
Hahah, I suppose we can leave this open for a possible solution? But one that would be an enhancement, since it reduces the number of steps?

hamishwillee, 2015-11-10:
If there is no cost/risk to sending a dummy movement command before the yaw command (on all vehciles) then yes, makes sense for us to do this in code and keep the issue open. I can't see any!

atomictom, 2015-11-12:
There is one issue. I was doing this by sending a send_global_velocity(0, 0, 0) call, but I found a bug where sometimes the drone hadn't reached the takeoff altitude before I sent the command. Unfortunately, that caused the drone to stop moving upwards. 

I currently use a variable `yaw_fixed` to only send the dummy command once (and reduce the chance of it messing with anything) and set it to True in any movement commands. When yaw is called it checks to see if yaw_fixed is True. If not, it issues the dummy movement command and sets it to True.

hamishwillee, 2015-11-13:
@atomictom Good point.**Any** command you send will replace the previous command on ardupilot and can potentially interrupt it. That is why the examples all use ``arm_and_takeoff`` with a synchronous wait for altitude being reached before allowing other commands. Downside of course is that you can't yaw or do anything else while taking off. 

I think the right way to fix this is to talk to @rmackay again about getting yaw to work during takeoff too. 

@tcr3dr I don't think we should add the dummy call in our code. We might be able to do this safely, but I suspect the code would be messy, less flexible, and a maintenance burden. If you're happy to settle for a takeoff that can't be interrupted, then this is much safer to implement in client code.

Thoughts?

atomictom, 2015-11-23:
@hamishwillee By the way, it's not just that yaw doesn't work during takeoff, but after takeoff too (until a movement command is sent).

hamishwillee, 2015-11-23:
@atomictom Yes, I realise that. IMO the comment I added covers both cases:

> After taking off, yaw commands are ignored until the first “movement” command has been received. 
  If you need to yaw immediately following takeoff then send a command to “move” to your current position.

Arguably I could be even more clear and remove the first bit - ie change to:

> Yaw commands are ignored until the first “movement” command has been received. 
  If you need to yaw immediately following takeoff then send a command to “move” to your current position.

Thoughts?

atomictom, 2015-11-30:
@hamishwillee Sorry, I was only referring to the "while" in your previous comment:
>Downside of course is that you can't yaw or do anything else while taking off.

and didn't think about looking at the documentation.

Actually, the first version is probably better because otherwise I think you'd want to mention that takeoff does not count as a movement command, but the first bit precludes that possibility.

Nevermind me! =P