dronekit/dronekit-python #286: subdomain for dronekit assets

mrpollo, 2015-08-13:
we are currently using the s3 url we should move to cloudfront and add a subdomain

currently:
http://dronekit-assets.s3-website-us-east-1.amazonaws.com

proposal:
http://assets.dronekit.io/

hamishwillee, 2015-08-14:
@mrpollo Looks good to me. Will the old URL work after you add the subdomain? Point being that I need to have links in the docs that will work moving forward.

tcr3dr, 2015-08-14:
@hamishwillee Old URL will continue to work!

hamishwillee, 2015-08-14:
@tcr3dr But the one we have live at the moment doesn't:
 https://s3.amazonaws.com/dronekit-installers/windows/dronekit-windows-latest.exe 

Are you going to fix that or do we need to patch with:
http://dronekit-assets.s3.amazonaws.com/installers/dronekit-windows-latest.exe

mrpollo, 2015-08-14:
@hamishwillee not sure how you got that one but thats never going to work, it needs a bucket-name behind s3, we can't fix that, we need to push a fix to docs then

hamishwillee, 2015-08-14:
@mrpollo I'll create a patch now.