dronekit/dronekit-python #294: Battery Vehicle Attribute is missing from API

franknitty69, 2015-08-15:
According to the documentation, the Battery Vehicle Attribute is exposed via vehicle.battery. when trying to read vehicle.battery, an error occurs stating that no such parameter (battery exists) in api.py.

The same issue occurs when trying to execute the vehicle_state.py example on the dronekit website.

hamishwillee, 2015-08-16:
Hi @franknitty69 

I just ran this on the latest version (1.5) and got the following output from vehicle_state.py (with Battery output):

```
Location: Location:lat=-35.363261,lon=149.1652308,alt=0.0,is_relative=False
Attitude: Attitude:pitch=-0.000983667210676,yaw=-0.122925661504,roll=-0.00108970480505
Velocity: [0.03, -0.04, 0.0]
GPS: GPSInfo:fix=3,num_sat=10
Groundspeed: 0.0
Airspeed: 0.0
Mount status: [None, None, None]
Battery: Battery:voltage=12.589,current=0.0,level=99
Rangefinder: Rangefinder: distance=0.0899999961257, voltage=0.00900000054389
Rangefinder distance: 0.0899999961257
Rangefinder voltage: 0.00900000054389
Mode: STABILIZE
```

What version of MAVProxy/DroneKit are you using and on what computer? What autopilot are you connecting to? Can you show me the output you get from vehicle_state.py

franknitty69, 2015-08-17:
I'm connecting to a 3DR Solo. I'm not sure of the version of DroneKit running on Solo. The autopilot is Pixhawk 2.

**Connected to solo**
[image]

**Dronekit Python Error**
[image]

**Python Script that i am trying to execute**
[image]

**I did notice that .mavinit.scr had a bunch of libraries unloaded, battery included. Could that be the reason for the error?**
[image]

hamishwillee, 2015-08-17:
One possibility is that Solo still contains an older version of DK - battery support went in at v1.2. Can you access the console and find out using ``pip list``?

@tcr3dr - do you happen to know what version Solo uses and whether it is OK to update it?

franknitty69, 2015-08-17:
Hamish,

pip or pip list results in an error.
[image]

hamishwillee, 2015-08-18:
@franknitty69 The version of DK on Solo is 1.1 - which pre-dates addition of the code to get the battery information. If you need to get this information, you can always using the mavlink message handler and capture the value yourself. There is an example of how to do this in [this pull request](https://github.com/dronekit/dronekit-python/pull/287/files) (not on live docs yet, but you can read the source). 

At the moment there is no supported path for updating the version of DK on Solo. This is a known issue, and is being worked on. I can't give you an ETA, but I suspect we're looking at months not days for the next update.

I'm therefore closing this issue.

closed 2015-08-18 by hamishwillee