dronekit/dronekit-python #295: Documentation: python modules

stephendade, 2015-08-16:
Subsequent to a discussion over at https://github.com/Dronecode/MAVProxy/issues/181.

The dronekit documentation should make it clear that, if the user uses the Windows installer, MAVProxy (and dronekit) will not have visibility of the user's default Python environment. Any extra Python modules they wish to use should be copied into the MAVProxy folder.

hamishwillee, 2015-08-17:
Thanks @stephendade . I've added the changes in #296 to address this.

tcr3dr, 2015-11-07:
@hamishwillee I think this is no longer the case now that we recommend ActivePython, yes?

hamishwillee, 2015-11-07:
@tcr3dr Yes and no. Yes this is no longer relevant, but it is because we no longer depend on MAVProxy in DroneKit, and are hence no longer dependent on its module loading behaviour.

We don't recommend ActivePython. I am still waiting on you to tell me if you would like me to do the necessary changes for that :-) I would want to do a bit more testing so this would be something that would need to wait a few weeks. 

Closing!

closed 2015-11-07 by hamishwillee