dronekit/dronekit-python #301: DK2.0 Documentation Changes Log

hamishwillee, 2015-08-17:
This is a log for tracking changes that will need to be made to the documents to support DK 2.0

* [ ] Examples, add ``from droneapi import connect`` and new connect syntax. Use an option parser to take in the connection, defaulting to SITL local port
* [ ] Document explaining changes and implications of changes vs how to use the API vs v1.x
  * [ ] How to import connect and start a connection
  * [ ] No MAVProxy so how to connect
  * [ ] Passing options to the script now just normal arguments
* [ ] System Overview, with diagrams

hamishwillee, 2015-08-17:
@tcr3dr @mrpollo A few DK2.0 questions you may be able to help with 

* The api object we get back from ``connect()`` - is it a singleton or can we have multiple versions? Or to put it another way, can I now support multiple vehicles in one script
* In examples I shouldn't hardcode the connection so I will need to pass in as an argument. By preference, do you think it is reasonable to default the value to by the SITL local connection or should I always force a connection to be specified?
* We will no longer need our windows installer for MAVProxy. So should we delete the old installer?
* Which of the dependencies are still needed? My feeling is that now we don't *need* mavproxy, then installation on windows becomes easy - just the droneapi and removing dateutil
* We showed connecting to MAVProxy in DK 1.x because that was the default that you had to have. SHould be move to Mission Planner, or have both?

tcr3dr, 2015-11-30:
@hamishwillee What's the status of this PR? What questions can I help answer?

hamishwillee, 2015-11-30:
@tcr3dr Nothing. Closing.

I removed the following from scope.  I'd quite like to have done  a programming guide, but in some ways it will be good to settle and see how people use the system. Re the "GCS" section, that would come when we switch to DK-SITL. Still some work there because it doesn't work to use TCP on multiple connections.
* [ ] Create section on using ground stations with DK. MAVProxy, MP or Tower ...
* [ ] System Overview, with diagrams
* [ ] Create Programming guide - #397

closed 2015-11-30 by hamishwillee