dronekit/dronekit-python #302: SerialException: device reports readiness to read but returned no data (device disconnected?)

samacumen, 2015-08-17:
Hello,

I am trying to connect Pixhawk with Raspberry Pi and using DroneApi for it. Everything is setup, and able to arm and disarm. However, problem is when I use the drone Api examply vehicle_state.py:

Please read below:

1) Execute the following command:
api start /home/pi/dronekit-python/examples/vehicle_state/vehicle_state.py

2) I receive an error everytime:
SerialException: device reports readiness to read but returned no data (device disconnected?)

3) The following output is produced (for your reference):

. . . 
Overriding RC channels for roll and yaw
 Current overrides are: {'1': 900, '4': 1000}
 Channel default values: {'1': 0, '3': 0, '2': 0, '5': 0, '4': 0, '7': 0, '6': 0, '8': 0}
 Cancelling override

Reset vehicle attributes/parameters and exit
Got MAVLink msg: COMMAND_ACK {command : 11, result : 0}
Got MAVLink msg: COMMAND_ACK {command : 400, result : 0}
Exception in APIThread-2: device reports readiness to read but returned no data (device disconnected?)
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/droneapi/module/api.py", line 333, in run
    self.fn()
  File "/usr/local/lib/python2.7/dist-packages/droneapi/module/api.py", line 625, in <lambda>
    APIThread(self, lambda: execfile(args[1], g), args[1])
  File "/home/pi/dronekit-python/examples/vehicle_state/vehicle_state.py", line 119, in <module>
    vehicle.parameters['THR_MIN']=130
  File "/usr/local/lib/python2.7/dist-packages/droneapi/module/api.py", line 35, in __setitem__
    self.__module.mpstate.functions.param_set(name, value)
  File "/usr/local/bin/mavproxy.py", line 215, in param_set
    return mpstate.mav_param.mavset(mpstate.master(), name, value, retries=retries)
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/mavparm.py", line 25, in mavset
    ack = mav.recv_match(type='PARAM_VALUE', blocking=False)
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 339, in recv_match
    m = self.recv_msg()
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 302, in recv_msg
    s = self.recv(n)
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 758, in recv
    ret = self.port.read(n)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 456, in read
    raise SerialException('device reports readiness to read but returned no data (device disconnected?)')
SerialException: device reports readiness to read but returned no data (device disconnected?)
DISARMED

Any help is appreciated.

Thanks,
Shyam

hamishwillee, 2015-08-19:
HI @samacumen

I don't know the answer to this, so I referred this to our [gitter feed](https://gitter.im/dronekit/dronekit-python) - which you can join!

The response was: "It looks like he’s using using the pixhawk’s telemetry port to the Pi’s GPIO ports and not USB (which is the correct way and documented here diydrones/ardupilot#2093). Not sure what’s up, but maybe start by confirming that the author turned off the serial port login on the Pi, which also sits on the pi's /dev/ttyAMA0, and has to be turned off for your Pixhawk to work right on the Pi.

So, as above, have you connected it to the USB port and set up otherwise as above?

akwatson, 2015-08-19:
I wasn't totally clear in my gitter chat now that I re-read it.  I meant that the author appears to be properly using the Pi's serial port to connect to the Pixhawk.  Connecting a Pi to the Pixhawk via USB is problematic in my experience.  
@samacumen I'm curious if you also get this same issue when opening a serial connection to the Pixhawk using mavutil vice DroneKit and doing I/O?

samacumen, 2015-08-19:
Hi @hamishwillee and @akwatson : Thank you for getting this problem to your attention. I would try to be specific in what I am trying to achieve.

1) I have connected Pixhawk to RaspberryPi: Connected Pixhawk's Telemetry 2 pins to RPi's GPIO pins - Tx, Tx, +Vcc, Gnd.

2) Everything is setup as from the following link (incl. the connection diagram). It works as described in: http://dev.ardupilot.com/wiki/companion-computers/raspberry-pi-via-mavlink/

3) I have "disabled" the "serial port login on the Pi" as given in the above link (the above link also refers to your dronekit-python community wiki instructions as it is):

4) The port on RpI autodetects it as: ttyAMA0.

I do not understand why this should occur, it is a simple out-of-the-box setup and I am stuck at this point to proceed further. Please let me know if you may need more information.

Thanks,
Shyam

akwatson, 2015-08-20:
Hey Samacumen, your setup looks good and my guess is that this issue is larger than your use of DroneKit.  Can you confirm whether you still get the serial port error when just using mavutil vice DroneKit to open and read data from the pixhawk you've got attached to the pi?  Try reading/writing to the device returned from mavutil.mavlink_connection(device="/dev/ttyAMA0", baud=115200) and see if you get the same serial port issue. 

One problem I found when using the Pi as a companion computer was the Pi's ability to supply enough current to USB devices.  For instance, when I was using both a Pixhawk and a GSM modem on the Pi's USB ports, it would overwhelm the Pi's ability to supply enough current and cause serial port issues.  Just be aware that the Pi (by default) only supplies 600 milliamps to the USB ports.  You can "amp up" the current supplied to the Pi's USB ports to 1.2 amps by editing the file /boot/config.txt and adding the line max_usb_current=1 to the end of the file.  Of course, make sure that the AC adapter that is powering your Pi (and by extension also powering everything on your Pi's USB ports) supplies enough current in the first place.  I want to say my wall plug supplies like 2 amps to my Pi, but I'm not at my bench to check.

samacumen, 2015-08-23:
@akwatson : Apologies for the delay in reply, was on a short vacation. Now that I am back, I want to get heads on to resolving this issue. Thanks for the details.

-> I am aware of the current issue on RPi's in my previous projects, esp. when using a WiFi adaptor. My RPi is powered up via an external adaptor supplying ~1.8 Amp (which I saw as one of the favorable adaptor for Pi). I use the latest Rpi model (rev 2).
-> I have edited the /boot/config file to add the max_usb_current=1, sync and reboot the Pi. With these two steps, current should not/ no more be a problem.

Question:
1) Could you tell me how to use to mavutil command. Is it to be executed on the command line on Pi. Could you pls. guide me through the steps, can't find the exact documentation on how to do that online.

2) Should I be powering via the the USB port or GPIO (on /dev/ttyAMA0) is good? Currently, I power up the Pixhawk using a USB port and RPi via the external power cable via micro USB.

Could you also replicate the issue on your part. If needed, I could provide the SD card image of my Pi. Really would like to get on with this as soon as possible ;(

Thanks buddy!

samacumen, 2015-08-30:
@akwatson : I was wondering if you could help you out regarding my comment posted previous week.

Thanks,
Shyam

akwatson, 2015-09-06:
Hey Samacumen.  I don't have my pixhawk to test with at the moment, but mavutil is just a utility class that comes with pymavlink.  In other words, you've already got it installed.  Try this and see if you're still getting the serial device error:  

from pymavlink import mavutil

AutopilotConnection = mavutil.mavlink_connection(device="/dev/ttyAMA0", baud=57600)
while 1:
    MavlinkMessage = AutopilotConnection.recv_msg()
    print MavlinkMessage

2.  For my bench setup, I power my Pi and Pixhawk via USB cables from a powered USB hub -- connecting only the ground, Rx, and Tx cable from the Pixhawk's telem2 to the pi's GPIO.  I've had other configurations work as well though.

3.  Please confirm whether you get the same serial error when launching dronekit/mavproxy using a USB  cable (not the telem2 to GPIO setup) to connect the pixhawk to the Pi.  This will place the pixhawk on /dev/ttyACM0 not /dev/ttyAMA0.

tcr3dr, 2015-11-07:
Hi, I am closing this issue due to inactivity. Please open up a new issue after upgrading to the latest version of DroneKit Python if you believe there still is a bug. Thanks!

closed 2015-11-07 by tcr3dr

cheintz, 2018-10-24:
I'm still seeing this issue using:

ArduPlane V3.6.0
Dronekit https://github.com/dronekit/dronekit-python/commit/a20eadf92b5d30940a7533a8a57a39273cdf3938
and  a Raspberry Pi 3b, connecting over usb to a Pixhawk PX4 (SR0)

peterbarker, 2018-10-26:
On Tue, 23 Oct 2018, cheintz wrote:

> ArduPlane V3.6.0
> Dronekit a20eadf
> and a Raspberry Pi 3b, connecting over usb to a Pixhawk PX4 (SR0)

Usually this is multiple things connected/reading from the one port.

suraj2596, 2019-12-22:
`sudo killall -9 mavproxy.py` : You might have run mavproxy before running the python script. In my case, after killing mavproxy, i was able to get data. This might be one reason.