dronekit/dronekit-python #303: timeout setting parameters via MAVProxy/Dronekit API

HuguesD, 2015-08-20:
Hi,
[image]
I'm executing a test.py script which is an exercise using basics of dronekit/MAVProxy and Wx to make a GUI, as shown on the attached picture.
I systematically get an error message when trying to set a parameter (as shown on picture). I also get these same timeout errors when using the official examples .py scripts (like vehicle_state.py).
I do not get these problems when I run the "Param set FS_BATT_ENABLE 2" on the MAvproxy prompt.
Any idea what is the cause ?  (note : all other communications with Pixhawk seem to work, like changing or reading a vehicle mode or reading parameters is also fine)

HuguesD, 2015-08-20:
apparently this is still an open bug, because eventhough "timeout" is displayed, the parameter is actually changed.
It is a false timeout message but what is really annoying is that this adds a delay of a few seconds for each parameter's write/set command !

hamishwillee, 2015-08-20:
Closing. This is a duplicate of #12.

The defect is a known issue that has been around for ages. We are expecting this to be fixed by DKPY 2.0 which brings a new architecture. This should be out in the "coming months". In the meantime all you can do is ignore the error.

closed 2015-08-20 by hamishwillee