dronekit/dronekit-python #304: Mathematical operations on objects

djnugent, 2015-09-01:
Implement __add__, __sub__, __mul__, etc to the Location and Attitude object. Simple but I am currently preoccupied by something else.

eliao, 2015-09-01:
I have a bunch of Location helper functions that could be put into Dronekit.

djnugent, 2015-09-01:
As do I. Probably not as much as you but I have made a fair amount of extensions to dronekit that should probably be standard...IMO

hamishwillee, 2015-09-02:
@djnugent What is the point? ie what does it mean to add, subtract, multiply two location objects? 

Same question  or Attitude - also at this point we can't set attitudes at this point (though that would be handy) so manipulating them doesn't seem useful.

It would however be useful to have more functions for manipulating postions/headings/distances. We already [documented a few frame conversion functions here](http://python.dronekit.io/guide/copter/guided_mode.html#frame-conversion-functions) and @mrpollo wants those ones in the API too (#133).

@eliao What are those functions you want to add? 

FYI from AC3.4 maybe not quite so necessary as copter will add [proper support for local frames](http://dev.ardupilot.com/wiki/copter-commands-in-guided-mode/#set_position_target_local_ned) so you will be able to specify more easily "distances X, Y, Z, metres from your current position).

eliao, 2015-09-02:
@hamishwillee The functions I had in mind include the frame conversion functions you mentioned.  
I do use a different method for calculating distance between two locations, and I have 2d and 3d versions (including altitude).  

newLocationFromAzimuthAndDistance is one I find super handy.  crossTrack functions may be useful as well.  

Once local frames are added, I expect one would want conversion functions between the two, as well as some vector math functions, which may be what @djnugent is referring to.

hamishwillee, 2015-09-02:
@eliao Your functions sound great. Minimally could we get them added to the documents? (ie what are they!) 

I'd like them in the API but we need to decide how. The easy way is to just provide a library of primitives. I like this because it means you can choose to use them or not. 

The hard way is to modify the Location object so that it understands frames better and handles conversions automatically. I like this because it reduces the things that users have to think about. I dislike it because all the methods have some level of error, and this approach makes it harder to swap out the options.

eliao, 2015-09-02:
Library of primitives sounds good to me.  

I think it should be fine to post my code, but I should make sure.  

Why are we posting code in the documents?  Is it a stopgap measure?

djnugent, 2015-09-02:
@hamishwillee I have just added a few things that make dronekit a bit easier to use...for me.  The biggest one is not letting the my script use location data, especially the home location, until we are armed. 

Adding objects means: `loc.lat = loc1.lat + loc2.lat; loc.lon = loc1.lon + loc2.lon; loc.alt = loc1.alt + loc2.alt;`
----> can be replaced with `loc = loc1 + loc2`

hamishwillee, 2015-09-03:
@eliao Great!

> Why are we posting code in the documents? Is it a stopgap measure?

Yes. Because there are lots of ways this can be done, I was hoping for a technical architect to mandate the approach that would be used (and ideally someone who has more Python experience than me to do the work). As I have the power to update the docs directly that was my "patch" on the problem. As it turns out, a good solution because no one has stepped forward to do the work ...

That said, I am sure if someone created a library that we could then discuss, it would be accepted unless the infrastructure was insane. hint hint!

@djnugent I see this useful for alts, but can you really just add two positions in *absolute* decimal degrees?

The home location change makes sense if all platforms set the home position only after arming. I'm not sure that is true - doesn't plane set it as soon as the GPS gets a lock? 

We're also planning some other work on home in #252 - worth adding a note about what you want to achieve in that post?

djnugent, 2015-09-03:
@hamishwillee no adding a relative to an absolute location. 

I'll give it a look.

hamishwillee, 2015-09-03:
@djnugent Thanks. As it is now the only "relative" information is the alt. It would be handy to add a location with values say in metres for lat, lon, but we would need some way for the system to manage relative frames and report misuse.