dronekit/dronekit-python #305: Expose the vehicle heading

hamishwillee, 2015-09-03:
This is last value in https://pixhawk.ethz.ch/mavlink/#GLOBAL_POSITION_INT

I think we can infer it from the Vehicle.velocity, but often people just want general direction.

eliao, 2015-09-03:
We'd need to make sure no one would confuse it for direction the vehicle is facing.  Can we make velocity a Vector3 object and then have a Vector3.direction function?

hamishwillee, 2015-09-04:
> We'd need to make sure no one would confuse it for direction the vehicle is facing. 

Agreed. I think that is a documentation issue. 

> Can we make velocity a Vector3 object and then have a Vector3.direction function?

I don't know anything about Vector3 and the costs/benefits of changing it from a list. My thought is that this would be useful if we needed to perform vector operations - and I imagine we will?

@mrpollo When you have time, would be good to have your feedback on this proposed change.

tcr3dr, 2015-11-07:
We currently expose the VFR_HUD property heading as `vehicle.heading`. Is this issue resolved?

hamishwillee, 2015-11-07:
@tcr3dr Since when? I never saw or documented this! Any other new ones you've added recently? I can see sys_status too, but I don't think anything else?

Yes this can be closed, but can you add a new issue to document it, and any other new attributes you added recently?

hamishwillee, 2015-11-09:
@tcr3dr So the changelist above adds docs for system status and heading. If there are any other new attributes (other than isarmable) let me know :-)

If not, IMO this can be set resolved.

tcr3dr, 2015-11-09:
@hamishwillee `.heading` snuck in during the Solo feature merge I think, heh. It is the only attribute I know of to get that treatment.

closed 2015-11-09 by tcr3dr

hamishwillee, 2015-11-09:
Great. We got an early request for this some time ago, so I'm sure our users will be very happy.