dronekit/dronekit-python #306: Vehicle.location returns wrong isRelative for copter

hamishwillee, 2015-09-07:
The location value has an isRelative field which is supposed to tell you if the altitude is relative to MSL or the home location. If it is true and you're sitting at the home location altitude should be zero. 

This is buggy for copter - if you look at the location value for (landed) Plane and Copter below you can see that the Copter value is broken - it should be True because we're sitting on the home position and altitude is zero.

* Copter: Location:lat=-35.3632507,lon=149.1652309,alt=0.0,is_relative=False
* Plane: Location:lat=-35.3632507,lon=149.1652309,alt=584.309997559,is_relative=False

ggregory8, 2015-09-10:
Is there a way to set what type of reference frame is used for altiude when retreiving vehicle.location. I.e I want to specifically receive AMSL.

Alternatively is there another way to retrieve AMSL altitude?

hamishwillee, 2015-09-11:
@ggregory8 Please don't hijack unrelated threads for questions - as it makes it hard to track the original problem. You can create your own issues for questions if you want, or perhaps read the answer I already posted in Gitter.

That said:

> Is there a way to set what type of reference frame is used for altitude when retrieving vehicle.location. I.e I want to specifically receive AMSL.

No. I requested this in https://github.com/dronekit/dronekit-python/issues/131.

> Alternatively is there another way to retrieve AMSL altitude?

Yes. All values in copter are returned relative to the home location, and its position is relative to the AMSL. So you can simply get the home location altitude (the guide shows how) and use it as an offset calculating the AMSL relative altitude on all other requests.

ggregory8, 2015-09-11:
Sorry I thought it was related! I guess #131 is it. THanks.

tcr3dr, 2015-11-12:
Deprioritizing from 2.0.0 pending more rigorous testing.

tcr3dr, 2015-11-24:
@hamishwillee I suppose we can close this? Are we good cross-vehicle now?

hamishwillee, 2015-11-24:
Yes!

closed 2015-11-24 by hamishwillee