dronekit/dronekit-python #309: Cannot connect outside of mavproxy

ssoRL, 2015-09-11:
Hi, I'd be very interested having <a href="https://github.com/dronekit/dronekit-python/issues/299">this</a> for the project that I'm working on, and decided to see if I could get it working on my system (Ubuntu 14.04). However when I try to connect to sitl, by running "droneapi.connect("127.0.0.1:4444")" on the python command prompt, the command prints "Started.", but then never returns (this seems to be happening somewhere in the MPFakeState prepare function). Any idea what would cause this?
currently on tcr-nomp branch
sitl started with sim_vehicle.sh --map --out=127.0.0.1:4444

hamishwillee, 2015-09-14:
@tcr3dr I'm not even getting this far - when I try and import "connect" I get `ImportError: No module named numpy`

jlian, 2015-09-15:
That's weird, numpy should come preinstalled,

hamishwillee, 2015-09-16:
> That's weird, numpy should come preinstalled,

Not pre-installed on Windows, though of course that might depend on the distribution.

It is weird though, because I don't get the error when connecting from a script.

hamishwillee, 2015-11-04:
I'm closing this. The issue was fixed, and numpy is no longer a dependency.

closed 2015-11-04 by hamishwillee