dronekit/dronekit-python #310: Guide: Better API Reference layout

djnugent, 2015-09-11:
Is there a better way to format the api reference page?  

Is it possible to have a more professional java doc looking layout? Or even a python looking doc like this http://docs.opencv.org/modules/highgui/doc/reading_and_writing_images_and_video.html

djnugent, 2015-09-11:
or like this https://docs.python.org/2/library/logging.html

hamishwillee, 2015-09-14:
HI @djnugent 

I think this could be a lot better too. The look&feel here is defined in the theme and is not actually DK-Python per-se - can you please [raise the request here](https://github.com/3drobotics/sphinx_3dr_theme/issues/)?

When you repost, can you please be more specific - "Professional" doesn't actually mean anything much to a UI designer. So looking at the themes you've pointed some concrete suggestions might be:

* Better highlighting of the main class level than just a horizontal rule - perhaps making the whole heading with a bar behind it.
* List the class structure in the sidebar as a table-of-contents