dronekit/dronekit-python #312: Add support for gimbal/mount

tcr3dr, 2015-09-15:
The Gimbal now is in a shipping product, and I have some early test code that shows how you would run it:

https://github.com/3drobotics/demo-realtimeearth/blob/master/example.py#L71

(I looped over the commands because I was over a very lossy UDP link)

We should integrate mount/gimbal commands and allow ROI and manual targeting.

hamishwillee, 2015-09-15:
Sounds great. For DKPY2.0? Are the Gimbal commands MAVLink compatible - ie standard and will work for any Gimbal?

tcr3dr, 2015-09-16:
@hamishwillee DKPY2 sounds fine by me. The Gimbal commands are MAVLink compatible and in fact found upstream in mavlink/mavlink. I have to *double check* this assertion is true on a message id level though.

hamishwillee, 2015-09-16:
@tcr3dr Thanks. Sounds excellent.

deb0ch, 2017-02-28:
DKPY2.0 is out already, and having tested it the other day the gimbal control works pretty well with dronekit, shouldn't this be closed ?

hamishwillee, 2017-03-01:
Sure. Gimbal control is here: http://python.dronekit.io/automodule.html?highlight=gimbal#dronekit.Gimbal

closed 2017-03-01 by hamishwillee