dronekit/dronekit-python #321: Ability to compare Location objects

wsilva32, 2015-09-25:
It would be great to be able to compare location objects or equality (or inequality).

Example:

a = Location(1.0,2.0,3.0)
b = Location(1.0,2.0,3.0)
>>> a = b
True

hamishwillee, 2015-09-28:
I'm not sure how useful that would be unless we could also add in a "tolerance" to the comparison. I can see a situation where you're at the same space within the tolerance of your GPS signal but the comparison still isn't matching exactly.

wsilva32, 2015-09-28:
Good point, a tolerance may be required for such an ability. It would be useful for performing unit testing for any code generating or reading location objects (rather than running comparisons on each lat,lon,alt individually). Unit testing does include an "assertAlmostEqual" but I'm not sure if that would work with object comparisons.

tcr3dr, 2015-09-28:
For the unit testing usecase, I support the change; in practice it's doubtful anyone would try equality otherwise.

Nominating for 2.0

hamishwillee, 2015-09-28:
Longer term this might need to change again. I'd quite like Location objects to respect a specified frame rather than return the boolean isRelative.