dronekit/dronekit-python #323: DKPY2: Arming motors message even when not arming

hamishwillee, 2015-09-30:
I've updated simple_goto.py (see branch tmp_getting_started_dkpy2 which will be merged in #320).

If I re-run the script when the vehicle is airborne the system still reports ">>> ☺ARMING MOTORS" etc. Yes, the script does set ``vehicle.armed = True`` but it is already armed and flying and the vehicle will ignore this command. So ideally in this case the message should not be shown.

```
Arming motors
Taking off!
 Altitude:  20.0100002289
Reached target altitude
Going to first point...
>>> ☺ARMING MOTORS
>>> ☺Initialising APM...
```

Note, I appreciate that perhaps this should just be ignored. I'm just raising the issue so we make a decision and I can document the behaviour. Ditto with ``Initialising APM``. Other alternatives 

- change the message so that it is clear that we're sending a command only
- remove all auto-generated messages - let the script author define their own messages if needed.

tcr3dr, 2015-11-07:
Since we can't change the message coming from APM, I would propose that we instead check if we are armed already. If we are, we do not set the arming command, thus preventing an excessive message from being sent.

hamishwillee, 2015-11-07:
@tcr3dr Good idea.

tcr3dr, 2015-11-11:
Fixed in #444.

closed 2015-11-11 by tcr3dr