dronekit/dronekit-python #324: DKPY2: GPS connection info not populated for a significant time after connecting

hamishwillee, 2015-09-30:
I've updated simple_goto.py (see branch tmp_getting_started_dkpy2 which will be merged in #320).

The script checks for GPS being good enough before attempting to arm. This check takes about 20 seconds to complete - even if the vehicle is already flying. As you can see in the example below the GPS connection information value is None - when it returns we get the current altitude as 20.

```
E:\deleteme\dronekit-python\examples\simple_goto>simple_goto.py
Connecting to vehicle on: 127.0.0.1:14550
>>> ☺Frame: QUAD
Basic pre-arm checks
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: 3
Arming motors
Taking off!
 Altitude:  20.0100002289
```

Obviously we're not getting a GPS message for the first 20 seconds after connecting (or we're somehow dropping it). Either way this needs to be fixed.

hamishwillee, 2015-09-30:
Note, testing this against a SITL master build - APM:Copter V3.4-dev (e0810c2e)

hamishwillee, 2015-10-01:
@tcr3dr This might actually be a problem with the build or the Copter version. Using dk-master build I get this on a windows native build (cygwin) of AC3.4-dev. I don't get it using ramon's pre-built vagrant build (AC 3.3rev1). I'll try another release on cygwin and see if its all versions or just this most recent one causing problems.

hamishwillee, 2015-10-01:
OK, I tried this on 3.3rc7 using Cygwin version with same result as above. 

I tried it with latest sitl-dronekit and 3.3rc11 (``dronekit-sitl copter-3.3-rc11 -I0 -S --model quad --home=-35.363261,149.165230,584,353``). As you can see below it waits for GPS about 8s. This is a bit more reasonable than 20s, but clearly "not right".

```
>simple_goto.py --connect tcp:127.0.0.1:5760
Connecting to vehicle on: tcp:127.0.0.1:5760
>>> ☺APM:Copter V3.3-rc11 (25f8c517)
>>> ☺Frame: QUAD
Basic pre-arm checks
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Waiting for GPS...: None
Arming motors
 Waiting for arming...
>>> ☺PreArm: RC not calibrated
```

tcr3dr, 2015-11-07:
Running on the #430 branch, my script looks like this:

```
python simple_goto.py --connect tcp:127.0.0.1:5760
Connecting to vehicle on: tcp:127.0.0.1:5760
>>> APM:Copter V3.3 (d6053245)
>>> Frame: QUAD
>>> Calibrating barometer
>>> Initialising APM...
>>> barometer calibration complete
>>> GROUND START
Basic pre-arm checks
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
Arming motors
 Waiting for arming...
 Waiting for arming...
 Waiting for arming...
 Waiting for arming...
Adds is_armable check.
>>> ARMING MOTORS
>>> GROUND START
 Waiting for arming...
>>> Link timeout, no heartbeat in last 5 seconds
>>> ...link restored.
>>> Initialising APM...
Taking off!
 Altitude:  0.0
```

Given the GPS + EKF + MODE checks, this seems like a reasonable amount of time. Would you agree @hamishwillee? You can also use the `--speedup 5` parameter in dronekit-sitl to make development faster.

hamishwillee, 2015-11-07:
@tcr3dr This is absolutely fine. In the problem case I am fairly sure that the problem was that DK was downloading parameters multiple times, and not actually getting the GPS messages. I haven't seen this problem for a while, so I suspect your new download/waiting code has fixed the problem.  Agree close.

tcr3dr, 2015-11-09:
Yaaay.

closed 2015-11-09 by tcr3dr