dronekit/dronekit-python #325: DKPY2: Exiting script does not send last command and may report exception

hamishwillee, 2015-09-30:
I've updated simple_goto.py (see branch tmp_getting_started_dkpy2 which will be merged in #320).

The last action in the script is RTL:
```
print "Returning to Launch"
vehicle.mode    = VehicleMode("RTL")
vehicle.flush()
```

However the RTL does not occur (the command is probably not sent - certainly the MAVProxy prompt was still reporting "guided".

I ran this a few times. In one case I got an exception reported:
```
 Altitude:  18.9200000763
Reached target altitude
Going to first point...
Going to second point...
Returning to Launch
Exception in thread Thread-1 (most likely raised during interpreter shutdown):
E:\deleteme\dronekit-python\examples\simple_goto>
```

Is there any way to ensure that the last command is fully completed (ie flush completes) before the thread exits?

Otherwise, what is the best way to deal with this? I'd prefer not to add documentation that scripts should wait a second before completing.

tcr3dr, 2015-10-14:
We need an explicit `vehicle.close()` command. This also addresses the "how do you properly terminate a script?" question.

hamishwillee, 2015-10-15:
Looks like that went in - see #347 and #348

hamishwillee, 2015-10-15:
@tcr3dr  I added `vehicle.close()` to end of simple_goto.py. The last sent RTL now completes, but I still get exception.

Last few lines of simple_goto.py:
```
# sleep so we can see the change in map
time.sleep(20)

print "Returning to Launch"
vehicle.mode    = VehicleMode("RTL")
vehicle.flush()

#Close vehicle object before exiting script
print "\nClose vehicle object"
vehicle.close()
```
Output is:
```
Returning to Launch

Close vehicle object
Exception in thread Thread-1 (most likely raised during interpreter shutdown):
```

tcr3dr, 2015-11-07:
@hamishwillee Can you check that this issue still occurs on rc8?

hamishwillee, 2015-11-09:
I can't replicate now after two or three tries. Lets "hope" this is fixed. Closing.

closed 2015-11-09 by hamishwillee