dronekit/dronekit-python #329: DKPY2: Remove RC override examples

tcr3dr, 2015-10-08:
I think it's appropriate to remove the RC override examples in DKPY2.

My primary motivation is not that this is unsafe user behavior (it can be but that's not relevant), but rather that this does not work on Solo, which has its own pipeline for RC controls. As a result, we don't have a generic way of this working even in the case of Python code, so we may want to approach this differently in the future.

For example, the `dronekit.sitl` package can include its own RC functions; `dronekit.solo` could do likewise. But Dronekit core would not include these.

tcr3dr, 2015-10-08:
Bonus: I don't have to re-implement the RC module, which is currently unimplemented.

hamishwillee, 2015-10-08:
@tcr3dr @mrpollo I disagree. This will work on most platforms and there is a standard way to do it. One device should not fragment the core so that you have to have an extra library for every variant.

Is it possible to define an API that would work for Solo and "generic MAVLink"?

If you don't want to implement this module for v2.0 could we still have the API definition and Vehicle.channel_override can return 'None". I will then document it as not supported in the current version.

hamishwillee, 2015-10-08:
or I can implement it ... if the module "pattern" you're using is obvious so I can copy it.

hamishwillee, 2015-10-08:
@tcr3dr Counter-argument? IMO this is something used by developers now for "good reasons". If its not going to work on Solo that is a pain, but we shouldn't hobble DKPY2.0 for other vehicles.

eliao, 2015-10-09:
I think the APM devs would heavily discourage any usage of RC Override.

hamishwillee, 2015-10-09:
@eliao They absolutely do encourage **mis-use**. The main use case is for the creation of virtual sticks, and you could reasonably argue that on a platform that is intended for Companion Computers it isn't needed. But ... some people do want to use them and there have been a couple of cases where I couldn't actually find an alternative way of doing something without them.

Chambana, 2015-10-09:
Regardless of whether the examples are pulled from the documentation, please leave the RC override capability in API for development purposes.

tcr3dr, 2015-10-09:
Agreed that these need to exist for development purposes and should not be removed.

hamishwillee, 2015-10-13:
#335 adds the rc-override code/docs back as a "standalone" example. It isn't tested yet but this can probably still be merged.

tcr3dr, 2015-11-07:
See #427.

closed 2015-11-07 by tcr3dr

hamishwillee, 2015-11-09:
Good. This was done. Will have to be done again for #427.