dronekit/dronekit-python #338: Fail to install dronekit-python on ubuntu 14.04 amd64

xros, 2015-10-13:
I followed the tutorial from here:
http://dev.ardupilot.com/wiki/droneapi-tutorial/

then I did this

# sudo pip install droneapi

errors were like this, hope someone can tell me how to fix this other than breaking my system.
>  
No local packages or download links found for python-gflags>=1.4

Traceback (most recent call last):

  File "<string>", line 17, in <module>

  File "/tmp/pip_build_root/protobuf/setup.py", line 200, in <module>

    "Protocol Buffers are Google's data interchange format.",

  File "/usr/lib/python2.7/distutils/core.py", line 111, in setup

    _setup_distribution = dist = klass(attrs)

  File "build/bdist.linux-x86_64/egg/setuptools/dist.py", line 262, in __init__

  File "build/bdist.linux-x86_64/egg/setuptools/dist.py", line 287, in fetch_build_eggs

  File "build/bdist.linux-x86_64/egg/pkg_resources.py", line 631, in resolve

    # Oops, the "best" so far conflicts with a dependency

  File "build/bdist.linux-x86_64/egg/pkg_resources.py", line 874, in best_match

    for key in self._distmap.keys():

  File "build/bdist.linux-x86_64/egg/pkg_resources.py", line 886, in obtain

    raise TypeError("Can't add %r to environment" % (other,))

  File "build/bdist.linux-x86_64/egg/setuptools/dist.py", line 338, in fetch_build_egg

  File "build/bdist.linux-x86_64/egg/setuptools/command/easy_install.py", line 607, in easy_install

distutils.errors.DistutilsError: Could not find suitable distribution for Requirement.parse('python-gflags>=1.4')

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip_build_root/protobuf

xros, 2015-10-13:
Oh I figured it out.

I should have used this repo and installed by my own.

When I was installing packages, I met tones of DNS, packages failures. 

Then I found it's because that I am in a region where is famous for its censorship all over the world.

I made it by using Tor.

closed 2015-10-13 by xros

tcr3dr, 2015-10-13:
@xros Happy to hear you got it working. Sad to hear packages were actively being blocked...

Helpfully, I went and analyzed where protobuf is being used and realized we can remove it as a dependency in the next version of DKPY. So this was a useful issue!

xros, 2015-10-13:
@tcr3dr I am truly happy if this is helpful. 
I am looking forwards the next version of DKPY. :smiley: 

P.S. I used Tor and dnscrypt to get rid of the blocking.
I only can fix the internet of my own. Just a little lonely in such a place.