dronekit/dronekit-python #342: DKPY2: Await attributes and parameters

tcr3dr, 2015-10-14:
Bikeshedding time~~~!

I feel like we want a method `.await([types], defaults=True)` that does the following:

1. Waits for system_status to enter an initialized state. This is equivalent to a `.notify_change` event on `"initialized"`. (included by default when defaults=True)
1. Waits for parameters to be loaded. This is equivalent to a `.notify_change` event on `"parameters"`. (included by default when defaults=True)
1. Waits for what other defaults? (Armed? Mode? etc.)
1. Waits for an arbitrary list of `.notify_change` events. This encompasses attributes.

So an example vehicle state example may be:

```
vehicle = connect('127.0.0.1:14550')
vehicle.await(['location', 'attitude', 'velocity', 'gps_0', 'groundspeed',
           'airspeed', 'mount_status', 'battery', 'rangefinder', 'mode', 'armed'])

print " Location: %s" % vehicle.location
print " Attitude: %s" % vehicle.attitude
print " Velocity: %s" % vehicle.velocity
print " GPS: %s" % vehicle.gps_0
print " Groundspeed: %s" % vehicle.groundspeed
print " Airspeed: %s" % vehicle.airspeed
print " Mount status: %s" % vehicle.mount_status
print " Battery: %s" % vehicle.battery
print " Rangefinder: %s" % vehicle.rangefinder
print " Rangefinder distance: %s" % vehicle.rangefinder.distance
print " Rangefinder voltage: %s" % vehicle.rangefinder.voltage
print " Mode: %s" % vehicle.mode.name    # settable
print " Armed: %s" % vehicle.armed    # settable
```

Wow, what a verbose function! But does it make sense? We want to *ensure* these values are populated. But this example then doesn't scale if we are always waiting for a rangefinder event that doesn't happen. Maybe we want to await a selection of always-on parameters, then time-await a selection of properties that may be optional?

```
vehicle = connect('127.0.0.1:14550')
vehicle.await(['location', 'attitude', 'velocity', 'gps_0', 'groundspeed',
           'airspeed', 'mode', 'armed'])

print " Location: %s" % vehicle.location
print " Attitude: %s" % vehicle.attitude
print " Velocity: %s" % vehicle.velocity
print " GPS: %s" % vehicle.gps_0
print " Groundspeed: %s" % vehicle.groundspeed
print " Airspeed: %s" % vehicle.airspeed
print " Mode: %s" % vehicle.mode.name    # settable
print " Armed: %s" % vehicle.armed    # settable

# optional params
time.sleep(3)

print " Mount status: %s" % vehicle.mount_status
print " Battery: %s" % vehicle.battery
print " Rangefinder: %s" % vehicle.rangefinder
print " Rangefinder distance: %s" % vehicle.rangefinder.distance
print " Rangefinder voltage: %s" % vehicle.rangefinder.voltage
```

Bikeshed away.

mrpollo, 2015-10-14:
I like this approach, we should also consider adding a timeout to await so it's useful on other "optional" params

```python
vehicle = connect('127.0.0.1:14550')
vehicle.await(['location', 'attitude', 'velocity', 'gps_0', 'groundspeed',
               'airspeed', 'mode', 'armed'])

print " Location: %s" % vehicle.location
print " Attitude: %s" % vehicle.attitude
print " Velocity: %s" % vehicle.velocity
print " GPS: %s" % vehicle.gps_0
print " Groundspeed: %s" % vehicle.groundspeed
print " Airspeed: %s" % vehicle.airspeed
print " Mode: %s" % vehicle.mode.name    # settable
print " Armed: %s" % vehicle.armed    # settable

# optional params
if vehicle.await(['mount_status', 'battery', 'rangefinder'], timeout=3, raise_exception=False):
  raise VehicleWaitException("timeout time exceeded")

print " Mount status: %s" % vehicle.mount_status
print " Battery: %s" % vehicle.battery
print " Rangefinder: %s" % vehicle.rangefinder
print " Rangefinder distance: %s" % vehicle.rangefinder.distance
print " Rangefinder voltage: %s" % vehicle.rangefinder.voltage
```

BTW the `raise_exception` portion is not mandatory, just trying to figure out how to fail if timeout time exceeded

hamishwillee, 2015-10-14:
My two bits:

1. Yes, we need something like this.
2. I would have had parameters as one of the things this awaits on - not something separate as per "2"
3. Like the idea of a timeout on pretty much everything - ie a choice.
4. Like the idea of being able to define some set of defaults
5. I like that you can await differently on mandatory/optional parameters

Thoughts/concerns

1. Vehicle.await "name" does not automatically make me think that this is a method that I have to call after connecting to allow parameters to accumulate.
1. Ideally this would be part of connect for that reason, but then you couldn't call the method multiple times. Perhaps it is part of connect and a separate function?
1. What if await is called at some point later in the script - would it fall through, or would it wait for the value to re-accumulate?
1. Is there some sort of "futures" approach you should/could use instead - so for example  a generic wait on a location != None future. I quite like the idea of futures because later on I'm probably going to want to wait on mode being a particular value etc.
1. As an aside, we also need to consider how we monitor the connection still being active - a different problem?

tcr3dr, 2015-11-03:
The PR #401 implements a method `vehicle.wait_ready(attr1, attr2, ..., timeout=30, raise_exception=True)`

From the initialization of the vehicle, we keep an ongoing list of what attributes have been emitted via the notification listener. wait_ready sees if those have been triggered, and if not, waits for them (up to a timeout, which can be None). For example:

1. `await_param` behavior is now expressed in terms of `wait_ready('gps_0', 'parameters')`.
2. If I have downloaded all parameters already, `wait_ready('parameters')` would return instantly.

etc. Note that certain attributes can become un-ready, or expire, such as "commands" and e.g. "link_alive" (if it were implemented). These are removed for the attribute record and we wait for the notification to be sent again in the loop. This seemed like the most intuitive behavior for invalidation.

We can also expose it as a parameter in connect(), if we want. I think the duplication is worth it. 

Idiomatically, we should also define what `connect(wait_ready=True)` does in terms of "default" values. My reasoning: we can standardize on `.commands.download(wait_ready=True)` to reduce the number of duplicate calls to download + wait_ready on the commands object. The semantics for the keyword argument would then be the same across objects and the whole codebase.

hamishwillee, 2015-11-03:
That meets all the points I had. 

With respect to mirroring in ``connect()`` and standardized idioms good idea. Not sure what the default set should be but minimally mode, armed, attitude. Possibly gps. This is the set that comes first (doesn't have to wait on anything like takeoff), and I am fairly sure will be output for every device. 

The good news is, if the defaults are wrong, we will easily be able to tune this by doing `wait_ready` afterwards on just a couple of vehicle-specific things.

tcr3dr, 2015-11-07:
Going to close this tracking issue as support has been landed (>= rc7).

closed 2015-11-07 by tcr3dr