dronekit/dronekit-python #343: DKPY2: gps_0

tcr3dr, 2015-10-14:
This looks like it was intended to be... different. There's also a bunch of FIXME's around this code. We want to access one or more GPS variables. Should this be refreshed?

hamishwillee, 2015-10-15:
Yes :-) 

In addition we also need to consider that the vehicle can have up to two GPS from AC3.2. The autopilot switches the primary GPS when one GPS has 2 or more satellites visible than the other. We need to confirm that what we're getting is the primary GPS information that matches up with out Location information.

If we decide that we only ever return the primary GPS then I'd prefer this attribute to be called .gps. If we're going to return both (assuming there are mavlink messages for this) and allow the autopilot to do something interesting with the second GPS information that would be another decision.

We should also see what other GPS messages are supported.

tcr3dr, 2015-11-07:
I think we can add this later, in a backwards-compatible way.

hamishwillee, 2015-11-07:
Fair enough. This is an enhancement we would like, but it can't effect too many vehicles yet!

atomictom, 2015-11-19:
I can see a use case for being able to access the second GPS for reporting/logging purposes.