dronekit/dronekit-python #348: Document Vehicle.close

hamishwillee, 2015-10-15:
@tcr3dr Can you explain the Vehicle.close semantics - when/why should this be called, and what exactly is the state of the vehicle after doing so? Is this called automatically when a script exits/falls through? etc.

hamishwillee, 2015-11-03:
This will be fixed by #384 (adds API reference for method). This is already documented in the migration guide.

tcr3dr, 2015-11-03:
Just changing this issue to note there needs to be a few more tests for vehicle.close() to ensure its behavior matches that in the docs.

hamishwillee, 2015-11-03:
Or update the docs to make sure they match what it does -  I more or less just documented what I thought it should do and was relying on you to catch me being wrong.