dronekit/dronekit-python #352: Mavlink callback handler only receiving parameters for first 20 seconds

hamishwillee, 2015-10-16:
The Mavlink callback handler (set using `set_mavlink_callback`) is receiving only PARAM_VALUE messages for the first 20 seconds (or so) after connecting, even when `await_params=True` has been set for the connection.

I've run some tests and confirmed that the parameter array is populated after the connection completes (just printed some values after the connection). I think this means that something in our startup code is possibly re-requesting these parameters.

This breaks the `create_attribute.py` example because for the first 20 seconds after connecting we don't get the messages we're searching for. Its also not "a good thing" :-)

This can be verified by modifying the create_attribute.py example as [per this gist](https://gist.github.com/hamishwillee/960fa8804c6406b26be7). Essentially it just logs the messages received, which you can see below are all params. I ran this with a native SITL build (master, copter) from Cygwin, and with the b6 DK build.

```
E:\deleteme\dronekit-python\examples\create_attribute>create_attribute.py
Connecting to vehicle on: 127.0.0.1:14550
>>> ☺APM:Copter V3.4-dev (e0810c2e)
>>> ☺Frame: QUAD
Param: 130.0
Param: 1900.0
Param: 1100.0
Display RAW_IMU messages for 5 seconds and then exit.
PARAM_VALUE {param_id : RC3_MIN, param_value : 1000.0, param_type : 4, param_count : 592, param_index : 80}
PARAM_VALUE {param_id : RC3_TRIM, param_value : 1500.0, param_type : 4, param_count : 592, param_index : 81}
PARAM_VALUE {param_id : RC3_MAX, param_value : 2000.0, param_type : 4, param_count : 592, param_index : 82}
PARAM_VALUE {param_id : RC3_REV, param_value : 1.0, param_type : 2, param_count : 592, param_index : 83}
PARAM_VALUE {param_id : RC3_DZ, param_value : 30.0, param_type : 4, param_count : 592, param_index : 84}
PARAM_VALUE {param_id : RC4_MIN, param_value : 1000.0, param_type : 4, param_count : 592, param_index : 85}
PARAM_VALUE {param_id : RC4_TRIM, param_value : 1500.0, param_type : 4, param_count : 592, param_index : 86}
PARAM_VALUE {param_id : RC4_MAX, param_value : 2000.0, param_type : 4, param_count : 592, param_index : 87}

...
...
```

As an aside if we're reading params from the vehicle on start, how do we know they are accurate if a GCS might also change the values? Do we re-read after updating a value ourselves (or do we update the array and assume the update was successful?)

hamishwillee, 2015-10-16:
Actually, interesting "factoid" is that the first twenty seconds of the ``guided_set_speed_yaw.py`` example look like this:
```
Waiting for GPS...: None
Waiting for GPS...: None
...
Waiting for GPS...: None
```
I'll bet that this is because the system is still downloading parameters so it can't update GPS and other attributes.. 

This is interesting because  I don't see this when using other SITL builds - e.g. dk-sitl for copter 3.3 doesn't do this. That makes me think that this is an artifact of using Cygwin or copter 3.4 ... though possibly triggered by something that we do that only affects those builds.

tcr3dr, 2015-11-03:
Our behavior in DKPY1 was to repeatedly download parameters continuously. I am unsure if that is the correct behavior, and this is what we currently implement:

1. In our MAVLink event loop, while the parameters set is not complete, we will first request a full list of parameters. Then given a certain threshold (50 params) after a timeout, we decide whether to request missing ones individually, or en masse by requesting them all again. This behavior is shaped to work well over UDP and is largely unnecessary over TCP.
2. Once the parameters set is full, we issue a `parameters` notification (as of #401) and the `.parameters` property has all of its keys defined.
3. Additional parameter changes are sent without prompting from the autopilot via PARAM_VALUE. There is no need to re-request all parameters, as we are seeing updates.

My understand of this may be flawed! But I think this is how the parameters were intended to be designed: http://qgroundcontrol.org/mavlink/parameter_protocol

hamishwillee, 2015-11-03:
Makes perfect sense to me. I've asked Lorenz to comment on the protocol interpretation. He is pretty busy, but we might get lucky ...

LorenzMeier, 2015-11-03:
This sounds a bit like a bandwidth management issue? On the highest level the behaviour sounds right. We don't have to request the full param list for PX4 as we have found a way to do clever caching. Would be nice to get Dronekit support in for PX4, which would also show you some of the tricks we do to speed up the init sequence.

tcr3dr, 2015-11-07:
@LorenzMeier Opened a tracking issue. https://github.com/dronekit/dronekit-python/issues/428

I think the loading-once-of-parameters is how I'd like to go forward. @hamishwillee is there any documentation which needs to be changed to reflect this behavior?

hamishwillee, 2015-11-09:
@tcr3dr Yes, though I'm still not certain what you mean by loading-once-of-parameters. I think you're confirming that you'll download all the parameters all at once, and the user can ``wait_ready`` on them? Once all params are downloaded we will update them individually based on PARAM messages from the vehicle. Correct?

If so, I'd add a similar a bit like that to the second paragraph of [this section](http://python.dronekit.io/guide/vehicle_state_and_parameters.html#parameters).

tcr3dr, 2015-11-09:
@hamishwillee To synchronize state, we start out with an empty list of parameters, and download all their values at once. Once every value is populated with a non-None value, we never have to request a parameter again, because all parameter *changes* are automatically sent to us by Pixhawk. So for the duration of the script we will be in sync with all parameter state.

Let me know if I can clarify anything else on that, or a new PR addressing the docs in that section!

closed 2015-11-09 by tcr3dr

hamishwillee, 2015-11-10:
The parameters download information was added as part of #443