dronekit/dronekit-python #355: DKPY2 Can't clear waypoints

hamishwillee, 2015-10-18:
The following test code returns and error that `AttributeError: MPFakeState instance has no attribute 'vehicle'`  when I try to access Vehicle.commands.clear. Actually downloading of commands and reading home location works fine.

Test code:

```
from dronekit import connect
import time
import math
from dronekit.lib import VehicleMode, Location, Command
from pymavlink import mavutil

#Set up option parsing to get connection string
import argparse  
parser = argparse.ArgumentParser(description='Print out vehicle state information. Connects to SITL on local PC by default.')
parser.add_argument('--connect', default='127.0.0.1:14550',
                   help="vehicle connection target. Default '127.0.0.1:14550'")
args = parser.parse_args()

# Connect to the Vehicle
print 'Connecting to vehicle on: %s' % args.connect
vehicle = connect(args.connect, await_params=True)

cmds = vehicle.commands
cmds.download()
cmds.wait_valid()

print "About to test clear"
vehicle.commands.clear()
vehicle.flush()
print "Clear succeeded"
```

Output of test code
```
About to test clear
Traceback (most recent call last):
  File "E:\deleteme\dronekit-python\examples\mission_basic\test_mission_clear.py", line 29, in <module>
    vehicle.commands.clear()
  File "build\bdist.win-amd64\egg\dronekit\module\api.py", line 95, in clear
AttributeError: MPFakeState instance has no attribute 'vehicle'
```

hamishwillee, 2015-10-21:
@tcr3dr When you have time, I've marked this as urgent because it breaks two examples.

tcr3dr, 2015-10-27:
See https://github.com/dronekit/dronekit-python/pull/374 for the fix.

hamishwillee, 2015-10-27:
@tcr3dr Thanks - agreed fixed. Unfortunately now we can't upload waypoints - see #376

closed 2015-10-27 by hamishwillee