dronekit/dronekit-python #368: DKPY2: flush() should be renamed

tcr3dr, 2015-10-23:
flush() is misleading because it only operates on commands. `vehicle.commands.upload()` would be a more correct name for this behavior.

The idea of a flush() or batching of message changes until they are ready is a useful idea, but not one which is supported (and potentially cannot be supported under MAVLink currently)

hamishwillee, 2015-10-23:
Thanks. That was news to me - all the docs and examples assume that flush is needed for any change - e.g. for a new mode etc. I'll correct that!

tcr3dr, 2015-10-23:
I was surprised too :) https://github.com/dronekit/dronekit-python/blob/1c40e27452aa5dd9d4e887e049348ea5bee90b24/dronekit/module/api.py#L139

hamishwillee, 2015-10-23:
So presumably we also have to do this for `clear`

tcr3dr, 2015-10-27:
@hamishwillee Yup. I think anything which statically modifies the command array... I think we want to improve this API rather than figuring out the conditions where we need to use flush().

hamishwillee, 2015-10-27:
@tcr3dr We typically take too long to get this sort of change in - better to document how it works now if that is a quick job than leave wrong for an indeterminate amount of time. In any case #372 fixes the docs issue.

With respect to improving the API I'm really not sure what a developer would want here. While I would love to be able to batch up as below and send there is too much state information involved - any of these can fail 
```
mode change to guided
arm
takeoff

-> sendall
```
Even with commands, I'm not really sure you lose anything much if we were to make every single add operation be sent immediately - certainly not for the companion computer case, and possibly not even for the GCS case.

Upshot, not so concerned about the flush case. Far more concerned about providing APIs that confirm success/you can wait on. In particular a nice takeoff that handles everything for you in the ideal case but otherwise reports what is blocking in a sensible way.

tcr3dr, 2015-11-02:
As of #384 this will be fixed.

tcr3dr, 2015-11-03:
Completed.

closed 2015-11-03 by tcr3dr