dronekit/dronekit-python #370: 'send_ned_velocity' has a large time-delay on my controlling pixhawk

PenguinCheng, 2015-10-25:
Hello everyone! I am using the function 'send_ned_velocity' in my quadrotor(pixhawk) control. However I find that this function costs several seconds before the real velocity reaching the expected velocity, which can not meed the need of our real time control. Could anybody help? Which function in DronekitAPI can be used in real time control? Thank you in advance!

hamishwillee, 2015-10-25:
Hi @PenguinCheng 

DroneKit just sends the message at first opportunity. How quickly the Copter gets the message depends on the physical link. The way that the vehicle responds to this message in terms of acceleration will depend on both the autopilot and the physical characteristics of the Copter - in other words it is "out of scope" for DroneKit itself.

I've posted in the [gitter channel](https://gitter.im/dronekit/dronekit-python) for confirmation.

PenguinCheng, 2015-10-26:
Hello@hamishwillee
Thank you for replay！We use a serial port as physical link , and the baud rate is 115200. We have not accessed the autopolit code because it's a difficult work. So I don't know the frequency that autopolit code responses to the command. It may be the cause if sending message at first opportunity.

hamishwillee, 2015-10-26:
As far as I know the message is sent immediately. The response on Gitter should tell us more about autopilot handling.

rmackay9, 2015-10-26:
@PenguinCheng, I don't see many limitations on it reaching it's top velocity. I think it would be best to post a dataflash log somewhere showing the slow behaviour.

tcr3dr, 2015-11-02:
@PenguinCheng et al, I'm closing this issue as I believe it would be better to submitting dataflash logs to gitter or filed as an issue on diydrones/ardupilot. Let me know if you believe this is not the case.

closed 2015-11-02 by tcr3dr

PenguinCheng, 2015-11-03:
@tcr3dr I'm sorry I have missed that dataflash logs because I have never used them before. However, I find out that large control period(one second) makes this behavior. If I solve this problem, I will give a feedback　to gitter. And it's okay to close this issue.