dronekit/dronekit-python #371: Something we don't monitor blocking arming on early startup

hamishwillee, 2015-10-26:
The normal "arm and takeoff" code (as per [simple_goto.py](https://github.com/dronekit/dronekit-python/blob/master/examples/simple_goto/simple_goto.py#L29) example) checks that state is not initialising and that we have a good GPS fix. Then it changes the mode to GUIDED and tries to arm the vehicle. 

On the Windows Copter 3.3 dronekit-sitl build this starts and launches if you've waited a bit or already connected to mission planner. However if you start the script shortly after the simulator it never arms. This means that there is something it is waiting on - if you call the command to ARM before this "thing" completes the device will never arm.

If you add a delay of around 8 seconds after the vehicle changes to guided before calling ARM then the vehicle will ARM. If you just call ARM in the loop while waiting for it to arm this also works. To me this again indicates that there is something that we need to wait on before calling ARM.

Any ideas what it might be? The "easy" solution for the example is to just set arm to true in the loop - but this is pretty crap - much better to identify whatever pre-arm condition is stopping us from proceeding.

Thoughts?

tcr3dr, 2015-11-03:
This is related to #406 and a question I posed here: https://gitter.im/diydrones/ardupilot?at=56390d46a530033014e3ef33

Basically armed fails due to `>> PreArm: EKF warmup` preventing arming from taking off. I'm trying to figure out which condition this is so we can wait on it.

(Note this does not trigger in Copter 3.3-rc5.)

hamishwillee, 2015-11-03:
Great, thanks. What do you mean by "does not trigger"?

tcr3dr, 2015-11-03:
Rather, this is not an issue on Copter 3.3-rc5 (for whatever reason). Not relevant, just a note to why I wasn't seeing it.

hamishwillee, 2015-11-03:
Presumably the check still occurs. If it isn't blocking us, that almost certainly means that our test code for arming isn't running until after this check ... so perhaps some of the waits (for parameters etc) are now longer. This sort of thing scares me - hidden potential race conditions.

tcr3dr, 2015-11-07:
See #430.

tcr3dr, 2015-11-11:
#430 has been merged, giving us a generic way to wait for arming.

closed 2015-11-11 by tcr3dr