dronekit/dronekit-python #379: Release notes need to be updated to 2.0.?c

hamishwillee, 2015-10-28:
@tcr3dr As it says on the subject line!

tcr3dr, 2015-11-03:
Quick notes.

* local_connect etc is now connect(), and invoked directly from the command line
* set_mavrx_callback is removed. Use `vehicle.on_message('*', obj)` etc.
* wait_init and wait_valid both renamed to wait_ready.
* home_location extracted from commands array
* .parameters[] array no longer blocks on access for parameters to download
* mount_status, ekf_ok added
* add_attribute_observer -> add_attribute_listener
* numpy is no longer a requirement

hamishwillee, 2015-11-05:
Some of these need to go into the migrating guide (maybe all). Some tracking:

* [ ]  local_connect etc is now connect(), and invoked directly from the command line
* [ ]  set_mavrx_callback is removed. Use `vehicle.on_message('*', obj)` etc.
* [ ]  wait_init and wait_valid both renamed to wait_ready.
* [ ]  home_location extracted from commands array
* [ ]  .parameters[] array no longer blocks on access for parameters to download
* [ ]  mount_status, ekf_ok added
* [ ] add_attribute_observer -> add_attribute_listener - done by #420
* [ ] numpy is no longer a requirement

hamishwillee, 2015-11-10:
Added ekf_ok, information about removal of numpy dependency, as part of #443

tcr3dr, 2015-11-24:
Integrated via #480.

closed 2015-11-24 by tcr3dr