dronekit/dronekit-python #386: ImportError: no module named dronekit

franknitty69, 2015-10-29:
in dronekit 2.0rc1, python scripts fail when executing the import statement *from dronekit import connect*.

**steps to reproduce**
1. upgrade dronekit via *sudo pip install dronekit --upgrade*
2. clone dronekit-python repository from github
3. run vehicle_state.py example via *python vehicle_state.py --connect:127.0.0.1:14550*

**result**
Traceback (most recent call last):
File "vehicle_state.py", line 9, in <module>
from dronekit import connect
ImportError: No module named dronekit

**expected result**
vehicle_state.py script runs displaying SITL vehicle state

hamishwillee, 2015-10-29:
@franknitty69 This works fine - just tried it.

The likely cause is a dronekit incompatibility with dronekit-sitl.

Please do:
```
sudo pip uninstall dronekit
sudo pip uninstall dronekit-sitl
pip list
```
Confirm that neither dronekit-sitl or dronekit are present in the list - repeat the above process until they are gone. 

Then do 
```
sudo pip install dronekit
```
And proceed as before ....

franknitty69, 2015-10-29:
I have uninstalled droneapi, dronekit and dronekit-sitl. confirmed via pip list. i then installed dronekit.

[image]

when i execute import dronekit from python or via an example script i get the same error.
[image]

hamishwillee, 2015-10-29:
Is it possible that you have multiple installations of Python, and the one you're running DroneKit with is not the one associated with this pip list?

franknitty69, 2015-10-30:
resolved by uninstalling all version of python and re-installing python 2.7.10

hamishwillee, 2015-10-30:
That was the cause then - phew. Thanks for confirming.

closed 2015-10-30 by hamishwillee

ElefLee, 2017-03-30:
Dear friends
 I am facing the same problem when I'm trying to run in the "Simulation" of Mission Planner the examples " "Basic Mission" and "Mission_Import_export" from: python.dronekit.io/examples/
here is what i'm getting in Python:
Python 3.5.3 (v3.5.3:1880cb95a742, Jan 16 2017, 16:02:32) [MSC v.1900 64 bit (AMD64)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> 
 RESTART: E:\My Documents\PROJECTS\PROJECT TARHAN\PYTHON\Upload Mission File.py 
Traceback (most recent call last):
  File "E:\My Documents\PROJECTS\PROJECT TARHAN\PYTHON\Upload Mission File.py", line 16, in <module>
    from dronekit import connect, Command
ImportError: No module named 'dronekit'
>>>

the same message arrears and when I'm running the example in MP "Simulation"
Could some one help me please! I tryed almost everything ...

peterbarker, 2017-03-30:
On Thu, 30 Mar 2017, ElefLee wrote:

Mission Planner's python scripting is does not use dronekit-python.

-- 
Peter Barker                          |   Programmer,Sysadmin,Geek.
pbarker@barker.dropbear.id.au	      |   You need a bigger hammer.
:: It's a hack! Expect underscores! - Nigel Williams

ElefLee, 2017-03-31:
Dear Peter thank you for your response to my question.

Since I'm not an experienced programmer, could you please provide me more details what is the meaning of:

"Mission Planner's python scripting does not use dronekit-python."

and what would you suggest on how I could overcome this problem?

Sincerely

Elef Leego

hamishwillee, 2017-03-31:
Hi Elef

What Peter is saying is that the scripting used in Mission Planner is NOT dronekit-python. So you're effectively asking us for support about a completely different product - just doesn't make sense :-)

If you want to use dronekit Python and our examples then you'll need to read our documentation on how to set up an environment: http://python.dronekit.io/

If you want to use Mission Planner scripting then you need to look at this doc http://ardupilot.org/planner/docs/using-python-scripts-in-mission-planner.html

CHeers
Hamish

ElefLee, 2017-04-07:
Dear hamishwillee
thank you for your kind response. What I'm trying to do, is to run the examples found in 3DR-DRONEKIT (for example):
http://python.dronekit.io/examples/mission_basic.html   and
http://python.dronekit.io/examples/mission_import_export.html
I am trying to simulate them using "SIMULATION" inside the MP, before try them in a real Copter model, 
using Mission Plamnner 1.3.45. Initially I used python 3.5 to write scripts, but now I wend back to 2.7. 
I've followed the instructions in:
http://ardupilot.org/dev/docs/sitl-native-on-windows.html#sitl-native-on-windows 
to install SITL on Windows 7. I have installed also Cygwin and Mavproxy. 
Since, I've been lost, facing always errors, could you please give me the way (or suggest me where can I find the instruction to follow) in order to be able to run the above referenced examples?
Thank you once again

hamishwillee, 2017-04-11:
Hi Elef,

Start from here: http://python.dronekit.io/examples/running_examples.html

Avoid using MP at all for now - just use MavProxy. After you have the examples working you can see about adding a second ground control station.

Regards
Hamish