dronekit/dronekit-python #390: Vehicle.commands.next is not writeable

hamishwillee, 2015-11-02:
In DKPY1.0 the next field on commands was writeable to skip a step in a command sequence. This does not appear to be the case in #384. The mission_basic.py in that pull request demonstrates this (ie the [skip here is ignored]().

hamishwillee, 2015-11-02:
I don't know what last_waypoint is, but this has changed the meaning of next!

Previously next meant "the waypoint you are heading towards". So initially this returns 0, but as soon as you start moving it reports next as "1" the first waypoint. When you reach the waypoint the system reports "Reached waypoint 1". In the current branch now it is reporting "next" as the waypoint you just went to - the last waypoint. This means that you get next reported as 0 when you're heading towards the first waypoint. 

In addition, the mission_basic.py example shows that vehicle.commands.next is still not settable - ie in dkpy 1.0 as soon as you changed the number this was written and the target changed ... now changing the value makes no difference at all

hamishwillee, 2015-11-05:
This is not fixed (rc7/today's master). The .next command is ignored - the test code is the example mission_export.py in the master codeline.

There is an associated issue #419 which also breaks this example.

tcr3dr, 2015-11-07:
Fixed by #423 (on rc8).

commit 19a817ad (2015-11-02): Corrects last_waypoint and .next setting. Closes #390.
commit 45ec6dea (2015-11-03): Corrects last_waypoint and .next setting. Closes #390.
closed 2015-11-07 by tcr3dr