dronekit/dronekit-python #391: vehicle.commands.next == 0

darkobozhinoski, 2015-11-02:
After installing dronekit 2.0, i try to run the examples in SITL (more specifically i run mission_basic.py), but as I see the adds_square_mission() does not send the necessary commands to the vehicle. Could you please provide feedback regarding this?

hamishwillee, 2015-11-02:
@darkobozhinoski There are all sorts of things "not right" about mission handling in DKPY2.0.0rc1. The good news is that @tcr3dr has just created #384 which fixes almost everything - and I've updated the mission examples for that today. Not sure when he plans on pushing that live, but I suspect days, not weeks.

The only outstanding issue with what I've tested so far is bug #390 - this basically prevents you from changing the command, though you can read it. I should also note that I could be doing something wrong, so waiting on response to that defect.

Upshot, this should be closed by #384

tcr3dr, 2015-11-06:
This is now live as rc8 and mission_basic.py should work as expected.

closed 2015-11-06 by tcr3dr