dronekit/dronekit-python #392: vehicle.home_location should be settable.

hamishwillee, 2015-11-02:
There is a message for [setting the home location](http://python.dronekit.io/guide/copter/guided_mode.html#setting-the-home-location). We should make this settable and cache the value.

tcr3dr, 2015-11-02:
This is added now in the tcr-home branch.

hamishwillee, 2015-11-02:
@tcr3dr Does not appear to work. [Test code is here](https://gist.github.com/hamishwillee/13bafbdbe07a094e0158)

Basically this gets the home location before downloading the waypoints. It then downloads the waypoints in a loop until it gets a non-zero home location (ie after home location is actually set), then it explicitly sets the home location, then it reads back the home location (after a short delay). 

I say this fails because the read-back home location does not match. 

In addition, the docs say

> "If the GPS values differ heavily from EKF values, setting this value will fail silently."

Why? The home location can be set to anything and shouldn't rely on EKF positioning should it?

The test code also has an attribute observer test. I'm not expecting that to work.

``` 
Home Location (before downloading waypoints): None
 Home Location (after downloading waypoints - but possibly before we have a fix): LocationGlobal:lat=0.0,lon=0.0,alt=0.0,is_relative=False
 Home Location (when we have a fix): LocationGlobal:lat=0.0,lon=0.0,alt=0.0,is_relative=False
 Home Location (when we have a fix): LocationGlobal:lat=0.0,lon=0.0,alt=0.0,is_relative=False
 Home Location (when we have a fix): LocationGlobal:lat=0.0,lon=0.0,alt=0.0,is_relative=False
 Home Location (when we have a fix): LocationGlobal:lat=0.0,lon=0.0,alt=0.0,is_relative=False
 Home Location (when we have a fix): LocationGlobal:lat=-35.3632621765,lon=149.165237427,alt=583.989990234,is_relative=False
 Home Location (after setting waypoints): LocationGlobal:lat=-35.3632621765,lon=149.165237427,alt=583.989990234,is_relative=False
 Home Location (after setting then downloading waypoints): LocationGlobal:lat=-35.3632621765,lon=149.165237427,alt=583.989990234,is_relative=False

Add home_location attribute observer
set new home location
 Home Location (after setting waypoints): LocationGlobal:lat=-35.3632621765,lon=149.165237427,alt=583.989990234,is_relative=False
 Wait 2s so callback invoked before observer removed
```

tcr3dr, 2015-11-03:
Thanks for the test repro, I will get to work on this.

> Why? The home location can be set to anything and shouldn't rely on EKF positioning should it?

Oh, I meant to provide a link. This is literally the case, and one reason why I had trouble implementing an effective test: https://github.com/diydrones/ardupilot/blob/b15bf3243ed085409f77d211c7795ec9bc0ad8cc/ArduCopter/commands_logic.cpp#L843

hamishwillee, 2015-11-03:
Thanks very much. 99% certain the problem will be that my test home location is too far away - so perhaps I should retest this.

hamishwillee, 2015-11-03:
OK, so FYI, the distance you can be from EKF origin is about 50km in Copter 3.3. This means my test was no good - so I will re-write it and let you know how it goes.

The reason for the EKF check is, according to Randy, that the navigation controllers work relative to the EKF origin. As the floating point values only have 7 digits of accuracy for very large distances from the origin the positioning accuracy gets low enough to make copter movement jittery (the accuracy numbers don't actually look that bad as Randy explained them, but the measured result is quite poor in terms of performance)

I still think that is unlikely to be as big a problem as having an RTL position that is too far to get to :-0. Wish this did not fail silently.

hamishwillee, 2015-11-03:
@tcr3dr I updated the [test gist](https://gist.github.com/hamishwillee/13bafbdbe07a094e0158) so that instead of setting a fake location it gets the current location and just changes the altitude to 222. This ensures that it is within the EKF distance, but still "detectable" from the original home location set by the autopilot.

What this shows is that the location is correctly sent to the vehicle, and if you download the waypoints again it will be read as such.

However if you just read the ``home_location`` attribute after updating, but before redownloading the commands, it will have the "old" value. I would have though you would cache the value you sent to the vehicle so it can immediately be read.

Since setting the value can arguably fail then forcing the user to do a re-read is probably the right thing to do ... but it feels wrong! So please just confirm what you think is best and how it is going to be!

Output is:

```
 Home Location (before downloading waypoints): None
 Home Location (after downloading waypoints - but possibly before we have a fix): LocationGlobal:lat=0.0,lon=0.0,alt=0.0,is_relative=False
 Home Location (when we have a fix): LocationGlobal:lat=0.0,lon=0.0,alt=0.0,is_relative=False
 Home Location (when we have a fix): LocationGlobal:lat=0.0,lon=0.0,alt=0.0,is_relative=False
 Home Location (when we have a fix): LocationGlobal:lat=0.0,lon=0.0,alt=0.0,is_relative=False
 Home Location (when we have a fix): LocationGlobal:lat=0.0,lon=0.0,alt=0.0,is_relative=False
 Home Location (when we have a fix): LocationGlobal:lat=-35.3632621765,lon=149.165237427,alt=583.989990234,is_relative=False
 Home Location (after setting waypoints): LocationGlobal:lat=-35.3632621765,lon=149.165237427,alt=583.989990234,is_relative=False
 Home Location (after setting then downloading waypoints): LocationGlobal:lat=-35.3632621765,lon=149.165237427,alt=222.0,is_relative=False
```

hamishwillee, 2015-11-03:
Also, I have found that you cannot set the home location until it has a valid value from the autopilot - if you do so you will either get a location with 0 values or the autopilot home location. This is something autopilot-side right?

There is another long-standing bug we should fix. In copter the home location ``is_relative=False`` correct. But for normal locations the value should be true. I believe this bug does not affect Plane, but could be wrong.

```
Global Location: LocationGlobal:lat=-35.363261,lon=149.1652299,alt=0.0,is_relative=False
```

hamishwillee, 2015-11-03:
#403 fixes the documentation to say exactly how this works now. I'll modify it if needed in line with your response.

tcr3dr, 2015-11-07:
See #429.

tcr3dr, 2015-11-09:
This property is now implemented and documented.

tcr3dr, 2015-11-09:
> There is another long-standing bug we should fix. In copter the home location is_relative=False correct. But for normal locations the value should be true. I believe this bug does not affect Plane, but could be wrong.

@hamishwillee What do you mean by "normal locations"? Trying to figure out how to solve #306.

closed 2015-11-09 by tcr3dr

hamishwillee, 2015-11-09:
@tcr3dr On copter the frame for the *home location" reported by the vehicle is in mean sea level (isRelative=False). EVERY other location (ie "normal locations" :-) ) is reported in the relative frame, relative to the home location (isRelative=True). Either our code is misreporting the value sent by the vehicle or it is wrong.

The situation is more complicated for Plane, and shows a weakness in the API. There is nothing to stop a Plan using terrain data and reporting altitude relative to the terrain frame. In that case what should isRelative be? (in practice I think much more rare to see this, but it is supported).

Does that anwer?