dronekit/dronekit-python #393: DKPY2:  autopilot baudrate locked to 115200, should be configurable

a7scenario, 2015-11-02:
In DroneKit Python ver. 2 (DKPY2), the connect() function uses mavutil's mavlink_connection function to open a serial connection to the autopilot.  The issue is that connect's call to mavlink_connection doesn't specify a baudrate and, therefore, mavlink_connection just uses its default baudrate of 115200.  If I'm not mistaken, this means there is no way to change the baudrate of the connection between a Companion Computer and an autopilot if you're using DKPY2.  It would be better if connect() added an input argument of baudrate=115200 that it relays to mavlink_connection, to allow for using different baudrates with your autopilot. 

    def connect(**ip**, await_params=False, status_printer=errprinter, vehicle_class=Vehicle, rate=4):
        state = MPFakeState(**mavutil.mavlink_connection(ip)**, vehicle_class=vehicle_class)
        state.status_printer = status_printer
        state.prepare(await_params=await_params, rate=rate)
        return state.vehicle

tcr3dr, 2015-11-02:
Hi @a7scenario,

I'll add this ASAP to `connect()`. Thanks for filing this issue.

tcr3dr, 2015-11-02:
We'll publish this in the next rc, coming shortly.

closed 2015-11-02 by tcr3dr

hamishwillee, 2015-11-03:
I thought you could just pass the baudrate to the string  ``connect('com14 --baudrate 15000')``. This will require docs - and a fix to the connection string options http://python.dronekit.io/guide/getting_started.html#connecting-to-a-vehicle

commit 4a81fb4e (2015-11-02): Adds baud parameter to connect(). Fixes #393.
commit 5f2dcd42 (2015-11-02): Merge pull request #394 from dronekit/tcr-baud

Adds baud parameter to connect(). Fixes #393.
PR #410 (2015-11-05): Add/fix docs on baud parameter, debugg - 
Fixes

#393 Add doc for baud parameter
#118 Fix up the debugging 

Fix auto-mode doc use of slice. 
Remove excluded methods from automodule generation as these are not longer present in init