dronekit/dronekit-python #395: Document baud property of connect()

tcr3dr, 2015-11-02:
See #394.

hamishwillee, 2015-11-03:
I thought you could just pass the baudrate to the string  ``connect('com14 --baudrate 15000')``. This will require docs - and a fix to the connection string options http://python.dronekit.io/guide/getting_started.html#connecting-to-a-vehicle

tcr3dr, 2015-11-03:
@hamishwillee If that is actually the case (oh gosh) I would prefer not to add the explicit property; I very much like the 'connection string information is all that is needed to reestablish the connection' approach, much like SQL connection strings, etc.

I can't test "/dev/tty.usbmodem1415 --baudrate 115200" yet though. I'll have to snag an APM from someone.

hamishwillee, 2015-11-03:
@tcr3dr Well, I just made the assumption because you told me "connection strings are exactly the same as you would pass to mavproxy". I say lets test. I kind of prefer the explicit way because the API shows you what i can do ... rather than having to guess.

tcr3dr, 2015-11-03:
@hamishwillee Oh, in which case I am assuming the --baudrate property was parsed as part of the argument protocol, and would need to be exposed in the way the `connect(baud=...)` property does it.

hamishwillee, 2015-11-03:
@tcr3dr What is  are rate,  status_printer and  vehicle_class parameters to `connect()`? Specifically what values

```
def connect(ip, await_params=False, status_printer=errprinter, vehicle_class=Vehicle, rate=4, baud=115200)
```
Specifically what do they "mean" and what values might you reasonably change them to?

tcr3dr, 2015-11-09:
@hamishwillee 

`status_printer` is a method of signature `def status_printer(txt)` which is used to print STATUS_TEXT sent to the vehicle [as well as other diagnostic information—note, not sure if I want to stick with this interpretation, so maybe we just say a STATUS_TEXT printer]. It defaults to `lambda text: sys.stderr.write('>>> ' + text + '\n')`. Passing `None` attaches no listener to STATUS_TEXT. You can pass any method you want to do your printing or logging. (Note: maybe this should be a logging method by default?)

`vehicle_class` defaults to Vehicle. It refers to the class that will be instantiated from the class mechanism. You can change it to any subclass of Vehicle. See #437.

`rate` is the refresh rate sent to `request_data_stream_send` ultimately. A (default) value of 4 means 4Hz updates. 10Hz means ten updates per second. etc...

hamishwillee, 2015-11-10:
The connect docs/additional params are documented as part of #443

tcr3dr, 2015-11-11:
#443 has been merged.

closed 2015-11-11 by tcr3dr