dronekit/dronekit-python #397: DKPY2. Create programming guide

hamishwillee, 2015-11-03:
A guide containing general advice for programmers on using the API - this would duplicate some material from the getting started (e.g. how to connect) but would also cover much of the material in the migrating guide (which will eventually be retired). For example, stuff like how to call close when exiting a script. 

Other things to add:

* [ ] Comments about style you should use
* [ ] Explanation of why position values that are uploaded then downloaded are different (rounding error). This is explained in mission_import_export.rst but would be better in a guide.

 (and please add your own)

tcr3dr, 2015-11-03:
One thing I'd like to make happen shortly is [yapf](https://github.com/google/yapf) over all our files.

hamishwillee, 2015-11-03:
Yapf is a new one on me - looks like a very good idea indeed - and would effectively define what our style guide is.

tcr3dr, 2015-11-11:
I intend to run YAPF before the final release candidate.

hamishwillee, 2015-11-11:
Good. Might be worth running it earlier.

I'm more interested for this guide on more specific guidance to the use of the API - for example, stuff like whether it makes sense generally to write your takeoff code to wait on observers or to poll on is_armed. Advice on defensive coding - so for example in a real app it probably makes sense to have an observer on the system_status and attempt to do something defensive if the status goes to critical. Ideas welcome.

tcr3dr, 2015-11-11:
Agreed. To start spitballing on a list:

1. connect in a try catch block for maximum info: https://github.com/dronekit/dronekit-python/issues/414#issuecomment-155658984
1. poll on is_armable
1. attach event listeners to react to state changes (`mode`, `armed` most importantly)
1. execute launch sequence
1. do your thing
1. close your connection at the end

hamishwillee, 2015-11-11:
Thanks. Good start. I would have said poll on mode/armed during takeoff sequence because it makes flow of code more obvious, and because you can't execute code based on an observer "not triggering" (which even with is_armable might potentially happen). However I agree that you would also react to state changes in an observer - but I think mostly those that are critical like the status. YOu might watch for mode in order to update a GUI on current mode. I guess you might watch armed in case someone disarms in flight - but both those seem to be edge cases, not ones we can provide general advice on.

Thoughts?

atomictom, 2015-11-12:
One thing to think about is the ordering of function calls. For example, I was having a problem where I was doing some reinitialization in the vehicle.arm observer (on disarm) that needed to be done before my startup procedure could be called again. My startup procedure would exit if the drone was already armed, but in the time between disarming and the arm callback being called the startup procedure could be executed again. I solved that by having the arm callback set a "ready" variable at the end, and exiting startup if that variable was not True (startup then sets the ready variable to False).

Also, I'm working in a multithreaded environment and I've had some issues lately with the drone not being in a state I assumed it would be in. It might be good to advocate explicit assumption checking in all functions. Something like:

```python
def modify_altitude(amount)
    global desired_altitude
    
    # Can only modify the altitude if it's armed and not landing
    if not vehicle.armed or vehicle.mode.name == "LAND":
        return False

    # Bounds checking...
    desired_altitude += amount
    goto(desired_altitude)
    return True
```

hamishwillee, 2015-11-12:
@atomictom  keeping state "right" is pretty tough when asynchronous systems. 

As in https://github.com/dronekit/dronekit-python/issues/397#issuecomment-155938601 above, my feeling is that callbacks really are best only be used for global monitoring and handling  of system status or logging. Any sort of initialisation/movement/startup should be done synchronously, checking for success before proceeding.

So in your case I'd have a global observer on LAND mode (or RTL) which turns either exits the script or returns it to a safe state. The re-initialisation code would be part of the start of a normal initialization cycle. Of course this may well be much easier to say than do :-)

Generally though, if you can do this it should simplify any checking you might otherwise need to do in each of your individual functions.

That said, I think it is very good advice "not to assume state" and to consider whether there are conditions under which a command will not work. Also to take care when assuming that callbacks will be called in a timely manner.

* Does any of the above thinking seem mad?
* Is there any way we could improve the architecture to reduce these issues?

Thanks very much!

mikerob, 2015-11-13:
> As in #397 (comment) above, my feeling is that callbacks really are best only be used for global monitoring and handling of system status or logging. Any sort of initialisation/movement/startup should be done synchronously, checking for success before proceeding.

One thing I'd suggest, that may get me burned as a heretic, is to subclass vehicle. 

Copters are cool, but they're not the only thing that APM supports. If copter-specific stuff could go in a subclass it might make it easier for people with other types.

hamishwillee, 2015-11-13:
@mikerob  Can you please create that https://github.com/dronekit/dronekit-python/issues/397#issuecomment-156290791 as a separate issue for discussion.

If you're a heretic then I'm one too.

hamishwillee, 2015-11-13:
@tcr3dr What about performance. There is a little discussion on gitter about memory consumption. We probably should make notes that they don't need to thrash the system in tight while loops - thoughts?

> Hi, I am running dronekit on raspberry pi. I have a message_listener which I want to continuously receive message from. How should I keep the program from exiting (a simple while loop uses 80% of pi's cpu which is not acceptable) Thank you

> for my red-balloon-project i put sleeps into the perl script so that it wouldn't eat all the CPU.

atomictom, 2015-11-19:
@hamishwillee I'm actually using dronekit within a (multithreaded) web server, where commands can come in at any time, so unfortunately, doing things synchronously is difficult. 

One option is to have a global lock that you hold while doing things to force synchronicity (and it can be non-blocking so other requests can just fail). However, some commands (e.g. land) should probably come through immediately regardless of whatever else is happening.

I did something similar to what you mentioned. Basically, it can be thought of as a simple state machine where most operations fail if the drone is not armed and above a minimum altitude, and startup fails if it's not initialized or has already been called (I lock it). Callbacks are used to change that state.

As far as improving the architecture, I've been think about these issues a lot recently. For synchronous scripts, I think something like promises or continuations would be really useful (rather than having to explicitly wait on things). However, for asynchronous systems like mine, I'm not sure either of those is the right solution. You would probably still need to verify state before every operation since something between when you issued the operation and when your promise or continuation was activated could have changed things.

I've been trying to learn more about ways of handling asynchronous systems (multithreading, callbacks, promises/futures, CPS, FRP, etc.), but unfortunately I still don't feel like I have a good grasp on it all. 

One thing I've been thinking of is having a state machine and using something like decorators to annotate the required state you must be in for a given operation to work (and possibly any state transitions), but I'm not sure what if anything DK could do to support that.

To echo you: does any of the above thinking seem mad? =)

hamishwillee, 2015-11-19:
@atomictom 

Agreed. I like the idea of continuations/futures but I don't think they help us much here. The fact is that even if only your script and the vehicle are in the MAVLink network you still need to verify your commands because the transport is not reliable, and the vehicle can refuse commands. If you add another GCS or possibly multiple scripts into the mess then things only get worse, since the vehicle can choose to change the state you just set under the hood.

Dronekit can't do much in the way of global locks because it isn't the thing that owns the resources - the vehicle decides what it will obey. You can simplify things by setting the vehicle up so that it will ONLY obey guided mode commands from the script. I wonder if that is a possible partial solution ...  we could advise that scripts set themselves as sole-controller in guided mode. 

> One thing I've been thinking of is having a state machine and using something like decorators to annotate the required state you must be in for a given operation to work (and possibly any state transitions), but I'm not sure what if anything DK could do to support that.

I think that would "help", and we already do everything you need to implement it - decorators.

> To echo you: does any of the above thinking seem mad? 

Crazy like a fox :-)

I'm not sure we're any further along for advice to users though other than "you can't assume that your commands are received or acted on by the vehicle without confirmation, and in a scenario where the vehicle can also receive commands from other sources (e.g. a GCS) you can't assume that any state you set will remain set. Best you can do is monitor that state has changed in response to your commands, and that is isn't changed unexpectedly outside of your control.

atomictom, 2015-11-24:
@hamishwillee All good points, but

>the vehicle decides what it will obey.

is probably the most important thing to remember!

>You can simplify things by setting the vehicle up so that it will ONLY obey guided mode commands from the script. I wonder if that is a possible partial solution ... we could advise that scripts set themselves as sole-controller in guided mode.

As far as I understand, this is only possible through firmware, correct?

As in https://github.com/dronekit/dronekit-python/issues/482, I think it might be a good idea to recommend the use of `on_attribute` inside some sort of initialization function so that a new vehicle or multiple vehicles can be setup if need be.

hamishwillee, 2015-11-24:
> As far as I understand, this is only possible through firmware, correct?

I was thinking of [SYSID_MYGCS](http://copter.ardupilot.com/wiki/arducopter-parameters/#my_ground_station_number_arducoptersysid_mygcs) but on closer inspection this only states that it restricts RC overrides, not other commands. So I don't think we can "lock" commands to a particular GCS.

hamishwillee, 2015-12-29:
This has been addressed by #505, which adds a section on developing, including best practice: http://python.dronekit.io/develop/best_practice.html

closed 2015-12-29 by hamishwillee